raspiblitz/raspiblitz #436: 30initHDD formatting the HDD without prompt+confirmation

fluidvoice, 2019-03-16:
> Not sure because the drive has been formatted by raspiblitz.. possibly I accidentally formatted it to NTFS or something when I set it up initially to sync blockchain on my desktop. it was a while ago now.
> 
> Any recommended file recovery tools for Ubuntu? I'm hoping the blitz hasn't written over the data and corrupted it.. there is a chance it has because it formatted the drive and added ssh folder + raspiblitz.conf file.
> 
> Maybe for future release it would be good to have a notification saying blockchain not found and need a Y/N confirmation before it formats the drive in case we put the chain in the wrong folder or set the wrong filesystem on the drive.

Prob not worth the time/trouble to try to fix it.  I would download the torrent again if you don't already have it and sync again.

I agree it should prompt.  @rootzoll see : https://github.com/rootzoll/raspiblitz/blob/e38c48e54536a6233a0062b99858d43685635d80/home.admin/10setupBlitz.sh#L256  It prob should not reformat the drive without prompting the user to confirm.

https://github.com/rootzoll/raspiblitz/blob/4ccdd8bb22a1a0805588cf994735275871a51d48/home.admin/30initHDD.sh#L78

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issues/434#issuecomment-473569138_

rootzoll, 2019-03-17:
TODO: The script should check the HDD and ...

- when its the wrong format .. ask to reformat in ext4 or cancel
- when its ext4 and not empty .. info on how to prepare a correct Blockchain HDD and ask if format or cancel

gijswijs, 2019-04-26:
I would like to ask for an extra confirmation as well.
I just accidentally formatted a HDD with a complete blockchain downloaded on it, and an LND wallet on it as well.

rootzoll, 2019-04-30:
Starting with v1.2 the RaspiBlitz will ask before formatting the HDD.
Closing this issue.

closed 2019-04-30 by rootzoll