raspiblitz/raspiblitz #437: Adding background backup script

rootzoll, 2019-03-17:
See suggested Script by @vindard :
https://github.com/rootzoll/raspiblitz/issues/427#issuecomment-473592711

rootzoll, 2019-04-02:
Take this into account for a backup script - once LND release >0.5.2-beta gets aout:
https://twitter.com/alexbosworth/status/1112857863393763329

rootzoll, 2019-04-02:
more details see https://twitter.com/alexbosworth/status/1112900454818570241

So the idea would be to monitor the "channel.backup" file from the background process (every 1min) and on changes in checksum backup automatically to the SD and if set to a remote location (like a dropbox - because its encrypted).

rootzoll, 2019-04-02:
See details on how Static Channel Backups work in this issue:
https://github.com/lightningnetwork/lnd/pull/2313

openoms, 2019-04-03:
So still need to be careful with this one. It basically force closes the channels on recovery, but with the funds from the last state. Even with backed up data, the channels will be gone. Funds in the channels are theoretically revovered  minus the closing fees and fees do fluctuate a lot.
https://twitter.com/irek_zie/status/1113027512303521792?s=19

I am looking forward to have that RAID going.

rootzoll, 2019-04-03:
Yepp. RAID should protect your channels and the SCB is to recover your funds in a worst case scenario. Thats the plan for the next release.

rootzoll, 2019-04-06:
Details on how to use SCB:
https://wiki.ion.radar.tech/lightning-technology/research/static-channel-backups

vindard, 2019-04-07:
@alexbosworth made a script that could be useful for this.

It uses `inotify` to monitor for changes and do a copy. He also has instructions for running it as a `systemd` service

Script: https://gist.github.com/alexbosworth/2c5e185aedbdac45a03655b709e255a3

fluidvoice, 2019-04-07:
I bumped into this tool that can do CLI upload of files to various cloud storage providers.  Have not used it yet, but worth a look to see if it makes this easier looks like it's built on top of rsync. Runs on Windows and various linux:
https://rclone.org/

rootzoll, 2019-04-09:
Check out: https://gist.github.com/vindard/e0cd3d41bb403a823f3b5002488e3f90

rootzoll, 2019-04-12:
As part of support for SCBs I am rewriting the 70initLND.sh script to python.
TODO: Check if setup of virtual env on building sd card works and gRPC scripts work.

fluidvoice, 2019-04-12:
> As part of support for SCBs I am rewriting the 70initLND.sh script to python.
> TODO: Check if setup of virtual env on building sd card works and gRPC scripts work.

Will it be doing something like ...

> It monitors lnd's channel.backup file for changes and uploads those changes to Dropbox when detected.

If so, will you use [rclone](https://github.com/ncw/rclone) so there is support for other cloud storage providers than only Dropbox?

rootzoll, 2019-04-12:
The rewrite of 70initLND.sh is to make it easier to "Restore a Wallet with SCB file" on a fresh setup.

In a very first version of the Channel Backup Script it will make a backup from HDD to sdcard (just in case). Which tool to make a claoud backup is to be determine - I will take a look at rclone. If cloud backup will be part of v1.2 it would still needs manual config (editing the raspiblitz.conf by temrminal) - integration in UI dialogs will come later on, once people confirm its working.

vindard, 2019-04-14:
Hey, so I actually looked at rclone a bit before writing my script with the Dropbox api.

It looks like a pretty comprehensive tool, but I didn't like that it's such a broad package that I couldn't personally vet and track all dependencies of. Since my node is pretty much a very hot wallet (with the unlock password stored on the node along with various access macaroons) I wasn't entirely comfortable including anything where I couldn't directly track the dependencies of.

Even for Dropbox alone, there's a bunch of SDKs that are available but that I chose not to use either for similar reasons. For now, I feel like the best security tradeoff for me personally is to be as minimal as possible which is why I'm ok with simply calling the Dropbox api endpoint directly from bash, and why I'd probably simply include new api endpoint calls for any other services I choose to include in my scripts.

fluidvoice, 2019-04-14:
Perhaps this is an argument for just compressing to a backup file and let the user copy it to whatever cloud provider they prefer.  That has ZERO dependencies including extra github scripts.

rootzoll, 2019-04-15:
@fluidvoice What files should ne zipped and password protected as part of the backup. As far as I grasped the SCB concept - its just one file that is already encrypted by the LND seed. Its also very small - so no compression needed. So its just the simple case of uploading one file to a cloud service or external server.

vindard, 2019-04-15:
> @fluidvoice What files should ne zipped and password protected as part of the backup. As far as I grasped the SCB concept - its just one file that is already encrypted by the LND seed. Its also very [...]

Yes that's how I understand it as well. Single file, encrypted, mine currently just a few kb's big

rootzoll, 2019-04-15:
OK - First implementation works. Every 1min the _background.sh task checks if the Channel Backup File has changed. If so a copy to the SD card to `/home/admin/.lnd/data/chain/[..]/channel.backup` is made. Then the script checks is additional Off-Site Backup Targets are set. At the moment the following two options are available (and/or):

**A) SCP Backup Target**

In the `raspiblitz.conf` the parameter `scpBackupTarget` can be set with the value formatted like `[USER]@[SERVER]:[DIRPATH-WITHOUT-ENDING-/]`. On that remote server the publickey of the RaspiBlitz root user needs to be part of the authorized keys - so that no password is needed for the background script to make the backup. 

**B) DropBox Backup Target**

In the `raspiblitz.conf` the parameter `dropboxBackupTarget` can be set to a DropBox Authtoken. See how to get that token here: https://gist.github.com/vindard/e0cd3d41bb403a823f3b5002488e3f90

For the v1.2 Release this Off-Site Backup options need to be activated manually by editing the raspiblitz config. So they are for the more deep down users. If they run OK - in the following versions it should be more easy to set these Offsite-Backups by menu. The local copy to the sd card will be active automatically on a update to v1.2

TODO: Add a script to make it easier to generate and get/copy the root pub key.
TODO: Test a bit more

rootzoll, 2019-04-15:
TODO: Implement/Test recover Wallet with ChannelBackup File.

rootzoll, 2019-04-28:
script `/home/admin/config.scripts/internet.sshpubkey.sh` helps on init, show and transfer ssh-pubkey

rootzoll, 2019-05-03:
OK - so with v1.2 on setup of a fresh RaspiBlitz user can say I want to recover an old wallet and the upload channel.backup + enter seed words. The channel.backup file will get activated when node is synced and the main menu is called for the first time. The itv takes around 20min until all channel-funds get recovered back to the on-chain wallet.

rootzoll, 2019-05-03:
So for Static Channel Backup the RaspiBlitz has now a automatic backup script including remote backup to Dropbox & SCP. And also the possibility to recover with a channel.backup file.

Closing this issue for the 1.2 release.

closed 2019-05-03 by rootzoll