raspiblitz/raspiblitz #441: Failed to set LN password. Not beeing ask to enter PW again.

LeoPeo7, 2019-03-18:
I am installing Raspblitz v1 on my Raspberry Pi.
When it asked for a LN Password, my two entered Passwords didn't match.
It didn't ask me to set the PW again and continued to scan the bitcoin blockchain to 100%.

Now I am maybe stuck at "Lightning Filtering Blockchain".
It shows me in Progress:  ?/567672  Please wait...

Do I need to start over again? Main Blockchain is synced.

fluidvoice, 2019-03-18:
Login via SSH then hit ctrl-c to get to a command prompt, then run
`/home/admin/70initLND.sh` 
what happens?
if more problems what is your output of running
`/home/admin/XXdebugLogs.sh`

LeoPeo7, 2019-03-18:
I corrupted my blockchain.
Fuck me. I am starting from 0.
I tried a few things but in the end my blockchain got corrupted. I am
redownloading.

Thanks anyway for your help

rootzoll, 2019-03-19:
@LeoPeo7 check v1.1 it should support you better with repairing a broken blockchain.

closed 2019-03-19 by rootzoll