raspiblitz/raspiblitz #443: cant cash out

eliddell1, 2019-03-18:
All my channels are closed but no matter which  of my wallet addresses i try to cash out to, i get the error:
 FAIL - not a valid address ()

I have tried sending to my exodus and to a binance account.  my satoshis are untouchable. and im sort of livid.

fluidvoice, 2019-03-18:
> All my channels are closed but no matter which of my wallet addresses i try to cash out to, i get the error:
> FAIL - not a valid address ()
> 
> I have tried sending to my exodus and to a binance account. my satoshis are untouchable. and im sort of livid.

Run this from the command line when you get this error to show what's going on...
`sudo tail -n 50 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log`

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll