raspiblitz/raspiblitz #445: RESEARCH: Monitor syslog for Errors

rootzoll, 2019-03-19:
Use `grep error -i -A5 /var/log/syslog` to check for any other errors happening and see how we can support the user in fixing these.

fluidvoice, 2019-03-19:
This made me think we could automate error log collection and reporting as well.  I saw LinuxMint guys were using this site termbin.com which makes it easy to upload log files and any text on the command line user netcat (nc) like the following.  It auto generates a URL that can then be put in the git issue:
admin@RaspiBlitz:/var/log $ `sudo grep -ai failed syslog | nc termbin.com 9999`
[https://termbin.com/h08r](https://termbin.com/h08r)
So a script such as /home/admin/XXuploadLogs.sh can be run to do this. 
Prob good not to abuse the service though with constant uploading of huge files. But this might help debugging.

rootzoll, 2019-03-19:
Thats a nice tool - have to see were integration makes sense.