raspiblitz/raspiblitz #448: Running v1.1 now.

codeoholic, 2019-03-19:
1. Lightning Filtering Blockchain. It still shows - Progress: ?/567850.
2. du: cannot access '/mnt/hdd/bitcoin/indexes/txindex/042742.log': No such file or directory

CommanderPoe, 2019-03-19:
> 1. Lightning Filtering Blockchain. It still shows - Progress: ?/567850.

It is always like this to me... take some minutes then its all goodie

rootzoll, 2019-03-19:
The scanning info is on the list to be improved for future releases.

@codeoholic did it worked after a waiting time? The `cannot access` is a bit unsual

fluidvoice, 2019-03-19:
> 1. Lightning Filtering Blockchain. It still shows - Progress: ?/567850.
> 2. du: cannot access '/mnt/hdd/bitcoin/indexes/txindex/042742.log': No such file or directory

Verify that LND is indeed running?... run:
`/home/admin/XXdebugLogs.sh`
Also run this and post the resulting URL:
`sudo tail -n 500 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`

frennkie, 2020-07-20:
@codeoholic closeing due to inactivity and old version. Please try 1.5 or upcoming 1.6 and check/report issues separately.

closed 2020-07-20 by frennkie