raspiblitz/raspiblitz #451: Using NAS to store the LN/Blockchain data

hopitas, 2019-03-20:
Hi

Would it be possible to store the HD data to an external NAS? It could be NFS mounted to the Raspi? What would be the drawbacks with this setup? What would happen in case of network failure etc?  Could the blockchain or LN get easily corrupted?

fluidvoice, 2019-03-20:
> Hi
> 
> Would it be possible to store the HD data to an external NAS? It could be NFS mounted to the Raspi? What would be the drawbacks with this setup? What would happen in case of network failure etc? Could the blockchain or LN get easily corrupted?

I wondered about this too.  I don't see why not but I've not looked into it much.  I read that the ethernet port is actually faster than the USB 2 ports so could it even improve performance? dunno.
Try it! or search the Pi and Raspbian forums.

hopitas, 2019-03-20:
I have once lost my ln bitcoins due to power outage and I don't have UPS on my NAS. What happens if NAS shuts down unexpected? On the other hand NAS has RAID5 compared to USB HDD. I could try it out anyway, but after LN gets backup support, then NAS seems more robust.

geco91, 2019-03-21:
I think its worth a try but its a very manual special setup that would classify as a standard setup. To protect LND data better I think the research on this issue is more easy to setup and hopefully part of future releases: https://github.com/rootzoll/raspiblitz/issues/329

DaReaLDeviL, 2019-03-22:
Why not spin this thing up... can you make a VM version of it like for ESXI oder Hyper-V? VMWare in the small version is free and you can do free backups with synology or qnap nas or even veeam. So is makes room for bigger private nodes with more security.

rootzoll, 2019-03-22:
@DaReaLDeviL so the idea would be running the RaspiBlitz completly on a NAS system like synology or qnap? Interesting idea.

DaReaLDeviL, 2019-03-22:
Qnap and Synology are the biggest platforms (as far as I know) for home and professional nas  with 3rd party app installer and automated update options. This should be huge on the user base (tech driven people) and ease of install with better security brought by the nas itself. From the concept it should be a no brainer to jump on it. From the tech side I can deliver my time and effort to test. I can’t code, but read some of it and whould help as far as I can.

rootzoll, 2019-03-25:
@DaReaLDeviL at least some info on how to deliver packaged software to QNap/Synology would be interesting. As far as I remember they have quite different processor architecture, so I dont think porting RaspiBlitz is not an easy task. Also can everbody release packages that can be installed from outside the company stores?

openoms, 2019-03-25:
Regarding  building a Bitcoin and Lightning node on a NAS (FreeNAS) this is an awesome guide: https://github.com/seth586/guides/blob/master/FreeNAS/README.md

Another option is - since I love to use the Odroid XU4 with RaspiBlitz-on-DietPi already - to test this: https://ameridroid.com/collections/cases/products/cloudshell2-nas-kit

Smiggel, 2019-05-17:
I Would like something like this as well. I would like to be able to mount a folder on my NAS to the Pi, so the Pi stores all the data on my NAS. Did it before with an other installation. So it is possible. I just don’t know how to set this up with RaspiBlitz.

hopitas, 2019-06-04:
I have been too busy to try this out, but it should work by just copying /mnt/hdd to NAS share and making smb mount to NAS on boot?

For example put this to /etc/fstab:

`
//192.168.100.x/share/raspiblitz/ /mnt/hdd cifs user,username="user",password="xxxx",iocharset=utf8,file_mode=0777,dir_mode=0777 0 0
`

I don't know if this mount will happen early enough on boot so that everything is ready when bitoind and lnd starts. Another problem might be upgrade, where you need to make the fstab linking again. Is there a single place/way to tell the raspiblitz where it seeks the external hdd? This might be a useful config feature?

rootzoll, 2019-06-05:
> Is there a single place/way to tell the raspiblitz where it seeks the external hdd? This might be a useful config feature?

Not at the moment. Maybe check if the is a way for Raspbian to set something on the sd card before boot - like the way you can set WIFI before first boot.

fluidvoice, 2019-06-07:
Try adding:

rootdelay=5
in /boot/cmdline.txt

and

sudo mount -a
in /etc/rc.local

fstab is for OS dependant drives. Prob not the place to put it.

Smiggel, 2020-01-11:
I just tried to do a setup with a folder on my NAS mounted on the pi as /mnt/hdd, but I get the message that there is not HDD attached during setup. How I can accomplish the storage of my data on my NAS?

frennkie, 2020-01-11:
I'm afk right now. But I would suggest you grep for `sda` in `/home/admin`.

Smiggel, 2020-01-11:
> I'm afk right now. But I would suggest you grep for `sda` in `/home/admin`.

Sorry, don't really understand what you mean here. :)

Smiggel, 2020-01-11:
I am wondering if it would be possible to symlink the bitcoin and lnd on the external HDD to a folder on my NAS. It's not ideal, but then atleast the blockchain, etc would be stored on the NAS and not on the HDD.

frennkie, 2020-01-11:
I understand what you are trying to do. Currently this is not supported. In order to accomplish this you would need to remove some of the existing checks.

So if you run the following command you will get a list of locations where scripts currently assume that a physical external hard drive is connected to the Pi. You would need to "bypass" these.

```
grep sda /home/admin/*
```

Smiggel, 2020-01-11:
> I understand what you are trying to do. Currently this is not supported. In order to accomplish this you would need to remove some of the existing checks.
>
> So if you run the following command you [...]

Ah ok. Now I get it. :-) Thanks! I will look into it.

frennkie, 2021-11-13:
@openoms @rootzoll I was asked about this feature again today... having a RaspiBlitz with only the SD Card and no USB storage. And the Blockchain (and everything else) would then be stored on a remote device (e.g. via NFS or SMB). What do think of this?

rootzoll, 2021-11-25:
It sounds like a very special edge case setup. Did anybody managed to setup something like this (just a bitcoind that stores blockchain data over NFS/SMB) manually and can share some performance data?