raspiblitz/raspiblitz #452: lnd keeps crashing after upgrading to v1.1

ncstdc, 2019-03-21:
burned the v1.1 image to the sd card and booted up. but the lnd server seems to keep crashing. 

XXdebugLogs - https://termbin.com/wlzo

lnd.log - https://termbin.com/3j80

fluidvoice, 2019-03-21:
From a quick search that EOF in the LND log looks like a corruption of the LND DB. 
Was it working fine before you upgraded to v1.1?

ncstdc, 2019-03-21:
It was working fine before the upgrade. Is it possible to fix the DB?

CommanderPoe, 2019-03-21:
> It was working fine before the upgrade. Is it possible to fix the DB?

Did u stop lnd before the upgrade? Mine got corrupted once that way cuz i had forgotten about it,

fluidvoice, 2019-03-21:
> It was working fine before the upgrade. Is it possible to fix the DB?

Perhaps.  To recover: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#1-recover-lnd-data

fluidvoice, 2019-03-21:
> burned the v1.1 image to the sd card and booted up. but the lnd server seems to keep crashing.
> 
> XXdebugLogs - https://termbin.com/wlzo
> 
> lnd.log - https://termbin.com/3j80

found similar EOF error here in other LND issue: https://github.com/lightningnetwork/lnd/issues/1315#issuecomment-394544363

fluidvoice, 2019-03-22:
> It was working fine before the upgrade. Is it possible to fix the DB?

The LND guys said it looks like an SD card problem. Can you flash a different SD card? Might be a bad one.

ncstdc, 2019-03-22:
Same issue with a different SD card.

fluidvoice, 2019-03-22:
> Same issue with a different SD card.
> 
> Tried the rescue script with no luck either.

no prayer in helping without logs.
system (dmesg) logs, bitcoind logs, lnd logs, output from running the rescue script and/or description of what "no luck" means.
What are you using to burn the SD card? Is 
It's not normal to get this EOF error when starting LND.  It seems like something must be corrupted.

fluidvoice, 2019-03-22:
> > It was working fine before the upgrade. Is it possible to fix the DB?
> 
> Did u stop lnd before the upgrade? Mine got corrupted once that way cuz i had forgotten about it,

I don't understand what this means.  The upgrade procedure is powering down the node to remove the SD card to reburn it. It's possible powering down corrupted it but not very likely:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-my-raspiblitz-after-version-098

ncstdc, 2019-03-23:
I used etcher to write the images. I tried 2 other SD cards and all got the same error when starting LND. I also renamed the `lnd` directory on the external hd and created a new wallet. It ran without issues for over an hour before I shut it down. So I don't believe the issue is the SD card.

Whenever LND restarts, it would crash within a minute or two, but after a reboot, it would run for several minutes before crashing. With that extra couple minutes, I was able to force close all channels and now just waiting for the txs to mature. Hopefully I will be able to run LND long enough to catch up and cash out the wallet and start over.

DaReaLDeviL, 2019-03-24:
I have a question about this or some other update problems... whould it help to do a 2 card update process until we are out of beta (early stage with lnd and all the nice new things)? So it will be like: card insert in raspi is 1.0 - new sd card is 1.1 and get insert. If every thing runs fine the 1.0 get the new 1.2 card when it's released. If there is any problem just put the old 1.0 card in. Is this possible/reasonable in the raspiblitz setup?

rootzoll, 2019-03-25:
@DaReaLDeviL you can do this 2 card update procedure right now. You can downgrade from 1.1 to 1.0 ... I think only the macaroon/TLS.cert maybe need a renewal. But if this will be possible for all updates in the future is not sure (thats why I dont want to promote that too much at the moment) - future upadte may need to migrate some HDD data that are after that not backwards compatible.

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll