raspiblitz/raspiblitz #454: REASEARCH: Add Loop-out service to RaspiBlitz

rootzoll, 2019-03-21:
Lightning Labs released the Aplha of Loop:
https://github.com/lightninglabs/loop

For example running Loop on the RaspBlitz it could be possible to monitor the channel balances in the background and automatically send funds to on-chain wallets (hardware wallet) to regain inbound capacity and store earnings in a cold-wallet.

At the moment its very experimental and you would need to replace LND on the raspiblitz with a fresh self-compiled version from the LND Github master. And the loopd services needs to be installed. 

Interesting to see would be:
- how much resources does the loopd service consumes on the RaspiBlitz
- what config parameter would a automatic loop-out script need
...

So if someone is getting loopd runnign with LND on a RaspiBlitz, please report/share your experiences.

-

openoms, 2019-03-31:
AFAIK you will need to build LND from source to be able to test this feature until the new release is out.

rootzoll, 2019-04-02:
I heard that LND 0.6 release is just around the corner. Shortly after that the raspiblitz v1.2 will be out - containing the LND 0.6 and I think then everybody can start playing/researching on how to best integrate it to the RaspiBlitz.

openoms, 2019-11-07:
Started experimenting with the Loop out service.
First of all found that I need a new channel even with my fairly well connected node because no routes were found (possibly due to the increased fees of LNBig).
The installation script is functional, but testing is ongoing:
https://github.com/openoms/raspiblitz-extras#loop-service

Started this new repo to help testing new features in a compatible way but as a separate extension package. Will aim to include the other extra features too.

openoms, 2019-11-07:
My first successful Loop out:
Note the max predicted fee of 25147 and final fee of 4XXX sats.
`$ loop out --channel XXXXXXXXXXXXX --amt 500000`                                                                                                                                 
```
Max swap fees for 500000 Loop Out: 25147                                                                                                                                                                    
CONTINUE SWAP? (y/n), expand fee detail (x): x                                                                                                                                                              
                                                                                                                                                                                                            
Max on-chain fee:                 13851                                                                                                                                                                     
Max off-chain swap routing fee:   10010                                                                                                                                                                     
Max off-chain prepay routing fee: 36                                                                                                                                                                        
Max swap fee:                     1250                                                                                                                                                                      
Max no show penalty:              1337                                                                                                                                                                      
CONTINUE SWAP? (y/n): y                                                                                                                                                                                     
Swap initiated                                                                                                                                                                                              
ID:           XXXXXXXXXXXXXXXXXXXXX                                                                                                                              
HTLC address: XXXXXXXXXXXXXXXXXXXXXXXXX   

Run `loop monitor` to monitor progress.                                                                                                                                                               [3/72]
```

`$ loop monitor`
```                                                                                                                                                                           
Note: offchain cost may report as 0 after loopd restart during swap         
 LOOP_OUT INITIATED 0.005 BTC - bc1xxxxxxxxxxxxxxxxxxxxxxxxxx 
 LOOP_OUT PREIMAGE_REVEALED 0.005 BTC - bc1xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
```

---
`$ loopd`                                                                       
```
XX:17:XX.XXX [INF] LOOPD: Loop out request received                                                                                                                                              
XX:17:XX.XXX [INF] LOOP: LoopOut 0.005 BTC to bc1xxxxxxxxxxxxx (channel: 0xxxxxxxx)                                                                                    
XX:17:XX.XXX [INF] LOOP: Initiating swap request at height 60XXXX                                                                                                                                
XX:17:XX.XXX [INF] LOOP: 4xxxx Htlc address: bc1xxxxxxxxxxxxxxxxxxxxxxxxxxxx                                                                                    
XX:17:XX.XXX [INF] LOOP: 4xxxx state Initiated                                                                                                                                                  
XX:17:XX.XXX [INF] LOOP: 4xxxxx Sending swap payment lnxxxxx
XX:17:XX.XXX [INF] LOOP: xxxxxx Sending prepayment lnbxxxxxxxxxxxxxxxx
XX:17:XX.XXX [INF] LOOP: xxxxxx Register conf ntfn for swap script on chain (hh=60XXXX)                                                                                                          
XX:17:XX.XXX [INF] LOOP: xxxxxx Checking preimage reveal height 60XXXX exceeded (height 60XXXX)                                                                                                  
XX:17:XX.XXX [INF] LOOP: xxxxxx Waiting for either htlc on-chain confirmation or  off-chain payment failure                                                                                      
XX:17:XX.XXX [INF] LNDC: Payment xxxxxxxxxxxxxxxxxxxxxxxxxxx completed                                                                                      
XX:17:XX.XXX [INF] LOOPD: Monitor request received                                                                                                                                               
XX:23:XX.XXX [INF] LOOP: Received block 60XXXX                                                                                                                                                   
[... cut ...]                           
XX:23:XX.XXX [INF] LOOP: xxxxxx Swap script confirmed on chain
XX:23:XX.XXX [INF] LOOP: xxxxxx Htlc tx xxxxxxxxxxxxxxxxxxxxxxxxx at height 60XXXXX                                                                       
XX:23:XX.XXX [INF] LOOP: xxxxxx Htlc value: 0.005 BTC
XX:23:XX.XXX [INF] LOOP: xxxxxx state PreimageRevealed
XX:23:XX.XXX [INF] LOOP: xxxxxx Sweep on chain HTLC to address bc1xxxxxxxxxxxxxxxx with fee 0.00004XXX BTC (tx xxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)
```

openoms, 2019-11-24:
Loop GUI is coming to RTL: https://twitter.com/Suheb__/status/1198671293366898688

Shall we trigger these couple of lines (https://github.com/openoms/raspiblitz-extras/blob/master/config.scripts/bonus.loop.sh) during the RTL install in the future or give it an entry to the menu?

rootzoll, 2019-11-25:
I think loop should also be possible withouth RTL - so having an extra menu option would be best. But when user installs RTL - we can also trigger loop install to give the GUI full function.

rootzoll, 2019-12-14:
Merged into v1,4 branch --> #906

openoms, 2019-12-17:
TODO: reinstall services or delete from raspiblitz.info on bootstrap

openoms, 2019-12-18:
> TODO: reinstall services or delete from raspiblitz.info on bootstrap

included in https://github.com/rootzoll/raspiblitz/pull/916

rootzoll, 2019-12-18:
@openoms after I installed loop and see the loopd running, the simple try to use the `loop` command results in`-bash: loop: command not found`

openoms, 2019-12-18:
I had trouble with the Go paths, which I think have been fixed in https://github.com/rootzoll/raspiblitz/pull/918/

No after testing the latest state everything is installed on the right path:

```
admin@RPi4test:~$ loop | grep -c loop
5
/usr/local/gocode/bin//loop
admin@RPi4test:~$ which go
/usr/local/go/bin//go
admin@RPi4test:~$ which lndconnect
/usr/local/gocode/bin//lndconnect
```

Are you at the latest commit?

openoms, 2019-12-18:
Just tested with the new scripts after deleting go and works.
```
./config.scripts/bonus.loop.sh off
sudo rm /usr/local/go
sudo rm /us/local/gocode

cd /home/admin/raspiblitz
git remote set-url origin https://github.com/rootzoll/raspiblitz.git
/home/admin/XXsyncScripts.sh v1.4 -clean
cd
blitz
```

openoms, 2019-12-18:
I think it needs a login again for the new path settings in `/etc/profile` to be active.

Restarting the node just for that would be a unnecesary.

Could add a `source /etc/profile/` in the `00settingsMenu.sh` too

rootzoll, 2019-12-18:
Yes reboot is a bit much - but the easiest way for now to get this done.

openoms, 2019-12-19:
Could only reboot when Go was not installed previously.

For example if Go was installed for lndconnect in a previous session the GOPATH will be already loaded systemwide from /etc/profile and loop will be available straight after install.

openoms, 2019-12-24:
The loopd.service freezes the provision script on first start because it is waiting for the lnd.service to start.

Suggesting to not restore loop on first boot to avoid this problem. It is an advanced function which can be activated from the menu later.

rootzoll, 2020-01-23:
OK looks good. Service is running and also is stable after recovery/update.
So I am closing this issue.

openoms, 2020-01-24:
Watching: https://github.com/lightninglabs/loop/releases/tag/v0.4.0-rc1.beta

rootzoll, 2020-01-25:
OK why not taking the update :)

rootzoll, 2020-01-26:
OK looks good - v1.4 will running now v0.4.0-rc1.beta. Closing issue.

closed 2020-01-26 by rootzoll