raspiblitz/raspiblitz #455: Avoid too much logging

mrmx, 2019-03-21:
My node stopped due to a bad log rotation/extra logging, reaching this:

**$ ls -lh /var/log/**

`
-rw-r----- 1 root adm  **4.5G** Nov  3 17:31 daemon.log 

-rw-r----- 1 root adm  **3.9G** Mar 12  2019 daemon.log.1
-rw-r----- 1 root adm  4.0K Nov  3 17:16 debug
-rw-r--r-- 1 root root 135K Mar  3  2019 dpkg.log
-rw-r----- 1 root adm  4.0K Nov  3 17:16 fail2ban.log
-rw-r----- 1 root adm  1.7K Mar 16  2019 fail2ban.log.1
-rw-r--r-- 1 root root  24K Feb 14  2019 faillog
-rw-r--r-- 1 root root    0 Nov 13  2018 fontconfig.log
-rw-r----- 1 root adm  2.4M Nov  3 17:31 kern.log
-rw-rw-r-- 1 root utmp 287K Nov  3 17:19 lastlog
drwx--x--x 2 root root 4.0K Nov 13  2018 lightdm
-rw-r----- 1 root adm  2.4M Nov  3 17:31 messages
drwxr-x--- 2 root adm  4.0K Aug 13  2018 samba
-rw-r----- 1 root adm  **1.3G** Nov  3 17:31 syslog
-rw-r----- 1 root adm  **1.2G** Mar 16  2019 syslog.1`

fluidvoice, 2019-03-21:
Are you on v1.0 or v1.1? there was a fix for this in 1.1

mrmx, 2019-03-21:
Manual FIX:

**$> nano /etc/logrotate.conf**

Sample:

```
# rotate log files weekly
daily

size 100M

# keep 1 days/weeks worth of backlogs
rotate 1

# create new (empty) log files after rotating old ones
create

# uncomment this if you want your log files compressed
compress
```
$> sudo logrotate /etc/logrotate*

mrmx, 2019-03-21:
> Are you on v1.0 or v1.1? there was a fix for this in 1.1

Yep v1.0 ;-)

geco91, 2019-03-21:
please update to version 1.1 which has a fix for that. recommend this issue for closing.

closed 2019-03-21 by mrmx