raspiblitz/raspiblitz #456: RTL Web Admin

amcitaly, 2019-03-22:
Hello community, 
do someone know this: if I activate the RTL Webinterface from the Raspiblitz menu, it is sufficient to look the RTL on the windows machine? under the same local network of course, thank you. Bz I cant reach the RTL on my pc
[image]

rootzoll, 2019-03-22:
The WebUI is served as a webpage within the local network. You can call it from a windows machine, just anter that address into a web browser. 

> Bz I cant reach the RTL on my pc

Is that the same machine you run the SSH terminal on?

fluidvoice, 2019-03-22:
login via SSH then ctrl-c to command line and run
`sudo netstat -pln | grep 3000`
`nc -z -v -w5 192.168.1.66 3000` 
Perhaps your router needs to have port 3000 forwarding turned on?

amcitaly, 2019-03-23:
Thank you guys is done! Is my mistake, in these days I have been trying set the RTL with the configuration file that I found in the project of Shahana and for sure I make a mess
[image]
Then the Raspi reconnize the folder as not empty so it can't set, I have cancel and is done now. Sorry I didn't see that there was the possibility to automatically set it up by the Raspiblitz menu on Services, is so easy 
[image]
Thank you very much 👍

closed 2019-03-23 by amcitaly