raspiblitz/raspiblitz #457: RPC PW

apierp1, 2019-03-23:
/mnt/hdd/bitcoin/bitcoin.conf stores rpcpassword in plain text by default. Is this avoidable?

apierp1, 2019-03-23:
So does lnd.conf

fluidvoice, 2019-03-24:
> /mnt/hdd/bitcoin/bitcoin.conf stores rpcpassword in plain text by default. Is this avoidable?

This is a Linux admin question, not related to Raspiblitz. The solution involves file privileges and user access.  What are you concerned about?  Is your LAN and machine not secure?

geco91, 2019-03-24:
The RPC password has to be stored in bitcoin.conf ... thats the normal way to set the RPC password for bitcoind. LND needs the password also to be able to connect to the bitcoind.

I recommend this issue to be closed.

closed 2019-03-25 by rootzoll