raspiblitz/raspiblitz #458: Issues with "send" after update to v1.1

gehrlein, 2019-03-24:
Since I've updated to v1.1 sending does not work reliable anymore. It takes a very long time (sometimes it times out after 1min and sometimes it will just not update forever). Most of the times the amount is not send. Some other times it seems that the transaction goes through but it will still not update the sending process.

fluidvoice, 2019-03-24:
@rootzoll the version of LND was not changed from 1.0 to 1.1 was it?

geco91, 2019-03-24:
There was no change in LND between version 1.0 and 1.1 - maybe some channels that gave good connection before are still down and not reconnected?

gehrlein, 2019-03-24:
mh after some restarts and some waiting time it seems to work again - probably the channels needed some time to update!

closed 2019-03-24 by gehrlein