raspiblitz/raspiblitz #459: Blockchain "base" torrent extremely slow to download

thelwyn, 2019-03-24:
While I usually can download the "update" torrent within a day or 2, it takes me ages to download the base one:
- download often not running at all
- always at very low speed <20kb/s

for the "update "one, download starts immediately and I sometimes get decent speed.

I'm in China, but regularly downloading torrents well seeded at 1Mb/s or even faster sometimes.

Are the "update" and "base" torrent seeded differently?

I've explicitly forwarded some ports to the raspiblitz as suggested in another ticket, no difference.

fluidvoice, 2019-03-24:
> Are the "update" and "base" torrent seeded differently?

yes. this is dependent upon torrent users of course.

geco91, 2019-03-24:
@thelwyn maybe torrent is not the best option. Maybe prepare the "COPY" option by installing bitcoin core on your laptop and sync there the blockchain with `txindex=1` set in config and let it sync up. Then start with a fresh flashed sd card and choose the `COPY` option and follow the instructions.

fluidvoice, 2019-03-24:
@geco91 regarding txindex... https://github.com/rootzoll/raspiblitz/issues/69#issuecomment-474417796

geco91, 2019-03-24:
v1.1 is still configured to need `txindex=1` - so if you copy a unindexed blockchain to the RaspiBlitz it will start indexing, which can take much time.

thelwyn, 2019-03-25:
> > Are the "update" and "base" torrent seeded differently?
> 
> yes. this is dependent upon torrent users of course.

Yes, sorry it was more of a rhetorical question, I was wondering why there would be less torrent users for the base than for the update. But after more reading I understand that this torrent is solely used by raspiblitz, it's not an "universal" blockchain torrent downloaded all over the internet?

But why people would keep seeding update and not base?

I don't know how to display more information from rtorrent in terminal so I can't check peers, but if people are regularly setting up a raspiblitz these days I'm suprised there is such a difference of sharing between the 2 torrents.

Earlier today I was at 600kb/s on the update, 5kb/s on the base.

If that really takes too long I will try as geco91 suggested. But torrent worked fine at my first install, back in January I think.

rootzoll, 2019-03-25:
Yes the torrents are just for RaspiBlitz. I am not sure why the base torrent is less seeded - maybe its based on internal queing of seeding software, that gives smaller files more priority on incomming requests.

thelwyn, 2019-03-26:
There is something else happening I think.

If I try to download the torrent from my PC (using KTorrent) , which is on the same network than the raspiblitz, it's downloading faster. Not blazing fast, but much faster than the Pi, and it's also downloading when the Pi is not downloading at all. Like right now 170kbit/s from a seeder in France on my PC, 0kb/s on the Pi.

mamar59, 2019-03-26:
Hello sharing is done well at home, I'm in France but I observed this afternoon a decline in sharing
[image]

thelwyn, 2019-03-27:
If I try to sudo rtorrent from the Pi, nothing show up and it says it can't read the ./rctorrent.rc file.

Would you know where is located the rtorrent config on the Raspiblitz?

escapedcat, 2019-03-27:
Just fyi
```
*** 1) Status Torrent 'blockchain':
            xxx / 224257.5 MB Rate:   0.0 /  41.0 KB Uploaded:     xx MB

*** 2) Status Torrent 'update':
            xxx / 40244.6 MB Rate:   0.0 /  60.4 KB Uploaded:     xx MB
```
Getting this at the moment from within China. It started way higher and now kinda bounces around that speed. I guess I'll just wait till it's finished.

thelwyn, 2019-03-27:
Interesting thanks, I never get that speed for base, but sometimes quite fast on update... mystery. on the other hand I'm not 100% of time in front of it, maybe it's speeding sometimes (but according to the progress, not really).

@escapedcat is your raspiblitz behind a VPN?

escapedcat, 2019-03-27:
@thelwyn no VPN or proxy.

Currently:
```

*** 1) Status Torrent 'blockchain':
            xxx / 224257.5 MB Rate:   0.0 / 116.7 KB Uploaded:     xx MB

*** 2) Status Torrent 'update':
            xxx / 40244.6 MB Rate:   0.0 / 108.6 KB Uploaded:     xx MB
```

thelwyn, 2019-03-27:
STOP TEASING me =)
Who's your ISP? China Telecom for me.

thelwyn, 2019-03-27:
Well at least I'm learning stuff.

So for those interested to see more details when downloading the torrent:
- Ctrl-C to exit from the progress display and come back to regular terminal
- screen -r  --> to attach the screen where rtorrent is actually running
There, should have one or 2 torrent running depending on the progress. Use the arrow up/down to select one (some *** appears vertically next to it)
Then press RIGHT arrow to go to "download" view for this torrent.

And from there I can see that even though I'm connected to the trackers that sees 14 seeders, I'm only connected to 1 at a time, don't know why. It says I could connect to max 50 so not a peer limit problem. If I open the torrent from other computer with ktorrent, it immediately starts downloading from 2 to 3 seeders.

I'll keep looking later.

escapedcat, 2019-03-28:
@thelwyn sorry, not exactly sure which ISP. Internet is somehow provided by the building management. Will stop the the teasing :P

thelwyn, 2019-03-28:
@escapedcat oh well... good for you anyway :) I don't have the problem on the other torrent, nor on another computer so I doubt it's an ISP issue anyway.

(PS: your professional website header links to http://localhost:8080/)

escapedcat, 2019-03-28:
@thelwyn weird! Good luck anyways. Thx for the hands up. I fixed the link. Site is pretty much outdated by now.

bitsam, 2019-04-10:
@thelwyn , thanks for the command lines. it was driving me crazy not knowing what is going on with the torrent. I just setup my first raspiblitz node. Well it's not quite there, as my Blockchain base torrent has slowed to a snail pace. 

My results from the rtorrent screen (cli) is below. Does this mean there are no seeders online out of 9 ? Sorry for the newbie question, I can't seem to find the help file for rtorrent.

 [U 4/0] [D 7/0] [H 0/32] [S 0/9/127488] [F 256/256]

thelwyn, 2019-04-11:
@bitsam Be strong! It took me almost 2 weeks to download the base this time, and now it's been doing the final sync for 2 days =D

I don't know what that means about the seeders. I would interpret it as "there are 9 seeders available but you're connected to none of them". The online documentation of rtorrent is a bit messy and very incomplete.

Yeah, I saw that port too and didn't take time to compare with what was in the doc, you're right. I don't know if that would change something anyway. It seems the most important would be to met the raspiblitz keep seeding after downloading the blockchain. I think there is another ticket opened about this already.

Did you try to open and redirect 6934 to your Raspiblitz? Does it improve anything.

bitsam, 2019-04-11:
Hey.

I found a wiki on rtorrent I'll send later. My Pi had to reboot and it
picked another random port on the next boot. Opening the correct port does
help up the download rate and for others it makes you visable.  I am now
downloading around 2MB/s and uploading at 1MB/s

thelwyn, 2019-04-11:
@bitsam yeah I saw this wiki, as well as another doc, they are both messy and/or incomplete, unfortunately.

Not sure opening the correct port is changing much though. I also saw some times where I was downloading at few Ko/s and 1h later at mb/s without configuring anything for the ports. And I first installed a raspiblitz in January, at that time I downloaded the torrent much quicker without any visible configuration difference.

Hopefully I don't have to download that blockchain too often =)

bitsam, 2019-04-11:
Here is the wiki I used for rtorrent commands. It really worked well in the debain/rasbian os instance in Raspiblitz. https://github.com/rakshasa/rtorrent/wiki/User-Guide

thelwyn, 2019-04-11:
@bitsam I agree with you that some hint about checking the actual port used and sugesting to configure the NAT accordingly would be a good start.

bitsam, 2019-04-26:
I was able to torrrent from my local seed node (windows server) to my raspiblitz however, I noticed that the torrents on the raspiblitz seem to be split into two separate jobs. When i exit the status view with the steps mentioned above:

Ctrl-C to exit from the progress display and come back to regular terminal
screen -r --

I now get the following:

^Cadmin@RaspiBlitz:~ $ screen -r -->
-bash: syntax error near unexpected token `newline'
admin@RaspiBlitz:~ $ screen -r --
There are several suitable screens on:
        23115.pts-0.RaspiBlitz  (26/04/19 10:08:46)     (Detached)
        3546.update     (26/04/19 08:10:33)     (Detached)
        3506.blockchain (26/04/19 08:10:30)     (Attached)
Type "screen [-d] -r [pid.]tty.host" to resume one of them.

When I go into each torrent individually, I can see they are running on different ports. I only noticed this as I was only able to connect to one torrent at a time from my local seed server.

Has something changed, as I did this about a week ago, and was able to see both torrents running in the same session (on the same port)

closed 2021-04-17 by rootzoll