raspiblitz/raspiblitz #460: LDC Display stays white despite running with the newest release v1.1

norisg, 2019-03-24:
Just flashed the new raspiblitz version v1.1 and bought the screen from the german shopping list but screen remains white after connecting it. After going in the folder LCD-show and executing sudo ./LCD35-show raspberry is rebooting but screen remains white. Couldn't find any related issues.

norisg, 2019-03-24:
This is the screen: https://www.amazon.de/gp/product/B06X191RX7/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1

fluidvoice, 2019-03-24:
> Just flashed the new raspiblitz version v1.1 and bought the screen from the german shopping list but screen remains white after connecting it. After going in the folder LCD-show and executing sudo ./LCD35-show raspberry is rebooting but screen remains white. Couldn't find any related issues.

What is shown on a monitor connected via HDMI when it is booting up?  
Which power supply are you using?

norisg, 2019-03-24:
> What is shown on a monitor connected via HDMI when it is booting up?

I have no HDMI cable here right now but I am connecting to the pi via ssh. 
>Which power supply and SD card are you using? What did you use to flash the SD card? 

I am using a power supply with 5V and 3.0 A. I flashed the sd card with etcher (64 GB sandisk) and the set up is working fine. I can start to setup my raspi (via ssh) but the screen remains white. 
I tried another power supply nothing changed. I also used another sd card (32 gb) same problem. 

The red light remains constant all time and the green light flashes around 20 s

geco91, 2019-03-24:
The LCD looks correct. But its the first report that it stays white. Could be something with the LCD or with the Raspiberry GPIO pins. If you have a second Raspberry to exchange it would make sense to switch it or send the LCD back to Amazon and request a new one.

fluidvoice, 2019-03-24:
Please post a picture of the back of your LCD display to verify version. Does it look like the picture below with "**XPT2046**" written on it?

So your SSH display is normal and it's just the LCD screen that remains white?
this is probably the wrong driver is being used. try the following via SSH 
(note: these steps are from the Raspiblitz **build_sdcard.sh** script)...

`cd /boot`
`sudo wget http://www.raspberrypiwiki.com/download/RPI-HD-35-INCH-TFT/dt-blob-For-3B-plus.bin`
`sudo mv dt-blob-For-3B-plus.bin dt-blob.bin`
Then add the following lines the **config.txt** file then reboot.
>   dtparam=spi=off
>   dtparam=i2c_arm=off
>   #. Set screen size and any overscan required
>   overscan_left=0
>   overscan_right=0
>   overscan_top=0
>   overscan_bottom=0
>   framebuffer_width=800
>   framebuffer_height=480
>   enable_dpi_lcd=1
>   display_default_lcd=1
>   dpi_group=2
>   dpi_mode=87
>   dpi_output_format=0x6f015
>   #. set up the size to 800x480
>   hdmi_timings=480 0 16 16 24 800 0 4 2 2 0 0 0 60 0 32000000 6
>   #.rotate screen
>   display_rotate=3
>   dtoverlay=i2c-gpio,i2c_gpio_scl=24,i2c_gpio_sda=23

[image]

norisg, 2019-03-27:
So I checked the part number. It says XPT2046
I did what you recommended but still only white screen. I connected the screen to the most far pin away from the usb port. I think thats right or ?

fluidvoice, 2019-03-28:
Yeah, there's really only one way to connect it such that it's using the pins at the far end (away from usb ports).  All I can think of it to try again from the start burning a new SD card. If it doesn't work perhaps your LCD is bad and you should return it?  hard to know if you don't have another Pi to test it on.

fluidvoice, 2019-03-30:
> So I checked the part number. It says XPT2046
> I did what you recommended but still only white screen. I connected the screen to the most far pin away from the usb port. I think thats right or ?

Also, I was just thinking. You really need to connect the HDMI cable so you can see what is shown during bootup.  The system could actually be stuck on something that's not related to the LCD - which might actually be working fine.

norisg, 2019-04-03:
Today I tried it with an HDMI cable and everything looks good on the screen, but still no picture on the small LCD screen. Is there something in particular I have to look for when booting?

fluidvoice, 2019-04-03:
picture of the screen might help ;-)

norisg, 2019-04-09:
Ok I ordered a new LCD from another company (Kuman 3,5") now it is working fine. But this LCD screen works only with RaspiBlitz operating system. I tried raspbian but there it get stuck during the booting screen (after restart every time on a new boot line). But at least screen is not white haha. Thank you for your help, I will close this issue. :)

closed 2019-04-09 by norisg

fluidvoice, 2019-04-09:
> Ok I ordered a new LCD from another company (Kuman 3,5") now it is working fine. But this LCD screen works only with RaspiBlitz operating system. I tried raspbian but there it get stuck during the [...]

Which one, this?... https://www.amazon.de/Kuman-Raspberry-Touchscreen-Resolution-Protective/dp/B07D4FHBNW/