raspiblitz/raspiblitz #462: Unable to resolve hostname on Update

rootzoll, 2019-03-25:
see https://twitter.com/bhavman2/status/1110247015244414976

check to change hostname of raspberry just before reboot to prevent those errors messages.

bhavman2, 2019-03-26:
Close to 24 hours into this issue. Status hasn't changed even after reboot. 

[image]

command line flashes "sudo: unable to resolve host bhavman2LND" at the bottom of the terminal window every 5 seconds.. any ideas?

CommanderPoe, 2019-03-26:
Raspi-Config hostname? then set the old name you have on V1.0 ? Or is it something related to bitcoind rpc username?

bhavman2, 2019-03-26:
Looks like v1.1 set my hostname from the node alias I had set in the previous version which included brackets. Removing the brackets from my hostname did the trick. Not sure why I didn't encounter this issue with v1.0 though.

rootzoll, 2019-03-28:
TODO: check to change hostname of raspberry just before reboot to prevent those errors messages.
TODO: sanatize hostnames on update/migration

rootzoll, 2019-04-04:
This should be fixed now for v1.2 release

commit e780cfbe (2019-04-02): fix #462
closed 2019-04-04 by rootzoll