raspiblitz/raspiblitz #464: Blockchain sync failure message although correct

rookey, 2019-03-27:
I've set up the latest version of RaspiBlitz and copied all blockchain data from my local node, but the RaspiBlitz setup indicates it as invalid - incorrectly imo.

Status messages:

```
Found data in /mnt/hdd/bitcoin/blocks
Found data in /mnt/hdd/bitcoin/chainstate
Found data in /mnt/hdd/bitcoin/indexes/txindex
FAIL: less then 1500 .ldb files (696) in /mnt/hdd/bitcoin/indexes/txindex (transfere seems invalid)
*********************************************
QUICK CHECK RESULT
*********************************************
FAIL -> DATA seems incomplete.

```

The txindex directory content is correct, however, an 1:1 copy of my local one.
Can I somehow bypass this incorrect check that results in a break of the startup procedure?

thanks!

rootzoll, 2019-03-28:
Had the blockchain you copied over from your compter `txindex=1` switched on in its `bitcoin.conf`?

rookey, 2019-03-29:
Yes - actually it didn't before, but the setup instructions ("Copying from another computer") indicated that's how this setting is supposed to be, so I changed it, triggering this extensive indexing process.

rootzoll, 2019-03-29:
> Could I also set txindex=0 on the raspiblitz without problems?

Thats the plan for a future release: https://github.com/rootzoll/raspiblitz/issues/227

Feel free to try early and report if you see any problems with going forward without txindex.

rookey, 2019-03-30:
awesome, thanks!

closed 2019-03-30 by rookey