raspiblitz/raspiblitz #465: "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

rookey, 2019-03-27:
My RaspiBlitz setup went fine so far (apart from the incorrect blockchain sync error message I've described in the other message). 

However, after unlocking my wallet, I receive the above error message.
`""transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"`
My node has a fixed IP through which my Raspi can be reached from the outside, this worked fine using a RaspiBolt setup before. Also, this fixed IP can be found in /mnt/hdd/raspiblitz.conf , so I'm puzzled where the above is coming from and how it can be fixed?

thanks!

rootzoll, 2019-03-28:
LND fails to start the RPC server. So the unlocking and other actions on the RaspiBlitz fail. 
Please go to terminal and run `./XXdebugLogs.sh` and post output here.

rookey, 2019-03-30:
Meanwhile I've been able to set up my node successfully, not sure why it happened before, I think I had another LND instance running at that time. Thanks for your help - seems to have been a problem merely caused by some incorrect handling by me!

closed 2019-03-30 by rookey