raspiblitz/raspiblitz #466: stuck at filtering blockchain v1.1

coinicarus, 2019-03-27:
the issue i experience starts with the screen below waiting for bitcoind to start 
[image]

then we move onto the following:
[image]
it never moves past this. i have tried the sudo shutdown - r now twice to no avail 
when i attempt to get the logs i receive the following error: 
[image]

I appreciate your assistance.

fluidvoice, 2019-03-28:
some are reporting running the script `./70initLND.sh` fixed this LND getting stuck.

coinicarus, 2019-03-28:
> some are reporting running the script `./70initLND.sh` fixed this LND getting stuck "filtering blockchain".

ok cool! i will try this :) i'm guessing run the script and restart the node?

coinicarus, 2019-03-28:
> > some are reporting running the script `./70initLND.sh` fixed this LND getting stuck "filtering blockchain".
> 
> ok cool! i will try this :) i'm guessing run the script and restart the node?

forget it i got it i think :) thank you again!

coinicarus, 2019-03-28:
so far so good! it rebooted and came up now it shows completing final sync 👍 
[image]

coinicarus, 2019-03-28:
officially up and running, thank you again! 
[image]

closed 2019-03-29 by coinicarus