raspiblitz/raspiblitz #470: how can i change my dns??

ssrpanel, 2019-03-28:
when the device run,it shows me can not ping 1.1.1.1 ,how can i change the dns 1.1.1.1 to 8.8.8.8

rootzoll, 2019-03-28:
You would need to change that in the scripts. 
00infoLCD.sh (line 76)
80scanLND.sh (line 38)

rootzoll, 2019-03-28:
TODO: make DNS a setting in the raspiblitz.conf and set by config.script
TODO: test internet/online with multiple addresses (1.1.1.1 & 8.8.8.8 & etc) - just if all are not reachable, conclude that device has no internet

ssrpanel, 2019-03-28:
Thank u very much , it works!!!hahahah

rootzoll, 2019-03-29:
:) I keep thisn issue open so I remember the TODOs for a future release.

conejoninja, 2019-04-03:
This happens to me too, for some reason I can not ping 1.1.1.1, but 8.8.8.8 is fine. It's specially painful when updating raspiblitz and forgetting to modify the files, as it will hang the sync/update process until a working DNS is set.

I was thinking about adding a second check (first check 1.1.1.1, if it fails, then try 8.8.8.8), but your idea of using a list of possible DNS is way better, unfortunately my knowledge of bash scripting is quite limited.

rootzoll, 2019-04-03:
@conejoninja this will be part of next release

TODO: make DNS a setting in the raspiblitz.conf and set by config.script
TODO: test internet/online with multiple addresses (1.1.1.1 & 8.8.8.8 & etc) - just if all are not reachable, conclude that device has no internet

rootzoll, 2019-04-09:
OK. There is now a config script to set a static DNS server on your own:

```
/home/admin/config.scripts/internet.dns.sh [DNSSEVER-IP]
```

TODO: Test local and on Recover/Update

rootzoll, 2019-04-17:
Works. Part of raspiBlitz v1.2 ... closing Issue.

closed 2019-04-17 by rootzoll