raspiblitz/raspiblitz #471: RTL unable to login Password Not Found In LND Config! V.1.1

PapaCoderNit, 2019-03-28:
Im on version 1.1 Im not able to login RTL 
i get the error msg  Password Not Found In LND Config!
1. i reset my rpc (b) password.
2. uninstalled and reinstalled RTL
[image]

rootzoll, 2019-03-29:
This is a RTL based question - I am not that deep into the RTL code. Can you post the issue over at: https://github.com/ShahanaFarooqui/RTLFullApplication

PapaCoderNit, 2019-03-29:
Done thank you

closed 2019-03-29 by PapaCoderNit

goodc0re, 2019-04-06:
Having the same issue.

rootzoll, 2019-04-12:
For visitors -- continue here: https://github.com/ShahanaFarooqui/RTLFullApplication/issues/7

ShahanaFarooqui, 2019-04-14:
Please redirect all RTL related issues to repo: https://github.com/ShahanaFarooqui/RTL/