raspiblitz/raspiblitz #473: Lightning account name?

Eddie2312, 2019-03-29:
I see there is an option to change the nodes name but I make a channel with someone my alias shows up as a string of numbers. 

How do I set my alias appearance on the network?

rootzoll, 2019-03-29:
The string of numbers is your unique nodeid. If a UI is displaying the nodeid or the alias is not part of the protocol and is based on the UI implementation. There is no way to force that.

Eddie2312, 2019-03-29:
So I can't fix it? How did other people do it? I see website names advertized on almost all channels. Is that a casa node feature?

rootzoll, 2019-03-29:
Your alias is set by RaspiBlitz in the LND config and LND is communicating it to the network. Maybe you can get more details on the LND Dev Slack: https://dev.lightning.community/overview why your node is seen different. If you find anything that can be optimized on the RaspiBlitz for that, let us know.

closed 2019-03-29 by rootzoll

Eddie2312, 2019-03-29:
Sorry, there's nothing wrong, the name only showed up after I had made a channel