raspiblitz/raspiblitz #474: ENHANCEMENT: detect bad power supply (low voltage and throttling)

fluidvoice, 2019-03-29:
Found a great article - [What's Wrong With The Raspberry Pi](https://ownyourbits.com/2019/02/02/whats-wrong-with-the-raspberry-pi/)
In it, one thing pointed out is the problem of inadequate power supply that is a VERY common problem (as in a large number of users have it but don't know it)... it's a "silent" problem that causes all kinds of different symptoms and malfunctioning from slow/throttled performance, corrupted SD cards, hang/lockup, and probably also corrupted blockchain DB files.  So as an enhancement I think we should try to detect this situation using the `vcgencmd` to expose data showing the problem. The question is when is the appropriate time to measure this?
[image]

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issues/424#issuecomment-478178382_

fluidvoice, 2019-03-30:
@openoms wrote about this power-related data corruption here https://github.com/rootzoll/raspiblitz/issues/368#issue-415058136 which supports the need to detect this under-voltage if possible.

openoms, 2019-03-30:
Good find there. I don`t have a Pi running atm, but would be curious to see what it shows. Do you have all normal results when testing your Pi?

Other thing to test for HDD wear and tear is smartmontools. This article got me started with the basics: https://www.howtoforge.com/tutorial/monitor-harddisk-with-smartmon-on-ubuntu/
But there can be even timed tests and email notications set up fro different paramaters.
Both of my USB HDD adapters worked with adding the `-d sat` parameter:
` sudo apt-get install smartmontools`
` sudo smartctl -i -d sat /dev/sda1`
` sudo smartctl -a -d sat /dev/sda1 `
gives a bunch of possibly useful metrics. Especially alarming if the error rates move from 0.

raumi75, 2019-03-31:
@fluidvoice Thank you. Do I understand this right? In a perfect setup you have 1.325 V always? My pi mostly measures 1.2 and only occasionally 1.3312 V

fluidvoice, 2019-03-31:
> @fluidvoice Thank you. Do I understand this right? In a perfect setup you have 1.325 V always? My pi mostly measures 1.2 and only occasionally 1.3312 V

That's how I interpret what was written.  I'll update when I've tested mine.

openoms, 2019-03-31:
Very interesting. Running this script: https://github.com/bamarni/pi64/issues/4#issuecomment-292707581
I got this on Raspbian, RPi 3 B+ without the HDD connected with a 3A micro USB PSU with a power button (not my usual setup) :
```
admin@RaspiBlitz:~ $ sudo ./test.sh
To stop simply press [ctrl]-[c]

./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
51.5'C 1400/ 600 MHz 9223372036854775807 1.3250V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
51.5'C  600/ 600 MHz 9223372036854775807 1.2V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
51.5'C  600/ 600 MHz 9223372036854775807 1.2V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
50.5'C  600/ 600 MHz 9223372036854775807 1.2V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
51.5'C  600/ 600 MHz 9223372036854775807 1.2V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
50.5'C  600/ 600 MHz 9223372036854775807 1.2V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
50.5'C  600/ 600 MHz 9223372036854775807 1.2V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
51.0'C  600/ 600 MHz 9223372036854775807 1.2V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
51.5'C  600/ 600 MHz 9223372036854775807 1.2V
./test.sh: line 108: printf: warning: 10000000000000000000: Numerical result out of range
51.5'C  600/ 600 MHz 9223372036854775807 1.2V
```
This suggests massive throttling and low voltage.
Will test with the other PSU-s I got.

openoms, 2019-03-31:
Made a gist for it: https://gist.github.com/openoms/997f56822a6256234e14ba1c6055c840
removed the command causing the error message.

Run with this line:
`wget https://gist.githubusercontent.com/openoms/997f56822a6256234e14ba1c6055c840/raw/6d7ac0fce0fea47b1cb8ddf3daf56a626fa9e3d6/test.sh && sudo chmod +x test.sh && sudo ./test.sh `

I am very curious of your results!

fluidvoice, 2019-03-31:
My results on the "better" of my two power supplies.  The other one was pretty much unusable.

To stop simply press [ctrl]-[c]

63.4'C 1400/1200 MHz 1.2375V
63.4'C 1400/1200 MHz 1.2375V
62.3'C 1400/1200 MHz 1.2375V
62.3'C 1400/1200 MHz 1.2375V
62.8'C 1400/1200 MHz 1.2375V
62.3'C 1400/ 600 MHz 1.2375V
62.3'C 1400/1200 MHz 1.2375V
62.8'C 1400/1200 MHz 1.2375V
62.3'C 1400/1200 MHz 1.2375V
62.3'C  600/ 600 MHz 1.2V
62.3'C 1400/1200 MHz 1.2375V
63.4'C 1400/1200 MHz 1.2375V
62.3'C 1400/1200 MHz 1.2375V
62.3'C 1400/1200 MHz 1.2375V
63.4'C 1400/1200 MHz 1.2375V
62.8'C  600/ 600 MHz 1.2375V
62.3'C 1400/1200 MHz 1.2375V
62.8'C 1400/1200 MHz 1.2375V
62.8'C 1400/1200 MHz 1.2375V

fluidvoice, 2019-03-31:
Experimenting with two different Pi3B+ I have. One has 3.5" LCD, the other not - and two different power supplies both *supposedly* 2.5A supplies.
**Note:** my setup is a bit different since I'm using (testing) w/DietPi and USB boot (not SD card).

$ `sudo vcgencmd get_config int`
arm_freq=1400
core_freq=250
gpu_freq=300
sdram_freq=450

PS#1 (the better of the two)
state: starting up bitcoind, LCD screen, USB flash boot, USB HDD connected, keyboard connected, HDMI connected
**volt=1.2625V** 
PS#1 
state: starting up bitcoind, **no LCD**, no keyboard, HDMI connected, USB flash boot, USB HDD connected
**volt=1.3375V** 

PS#2 (would only power Pi+LCD with HDD connected if I waited to plug it in after boot started)
LCD screen, USB flash boot, no keyboard, HDMI connected
**volt=1.2000V** starting up bitcoind, 
$ sudo vcgencmd get_config int
arm_freq=1400
core_freq=250
gpu_freq=300
sdram_freq=450

fluidvoice, 2019-03-31:
@openoms 
I would love to know these numbers for the recommended "official" Rasberry Pi 2.5A power supply.  Particularly with an LCD and HD connected.  Can it handle them both?  Does it cause throttling?
I wonder at what voltage point does the risk of data corruption go up?

fluidvoice, 2019-03-31:
The comments in that script are informative:

> # Watch your Pi! RPi foundation knows that Micro USB for DC-IN
> # is sh*t but also doesn't give a sh*t. Voltage drops caused by
> # average USB cables cause all sorts of instabilities and also
> # data corruption but instead of fixing the problem they masked
> # it in their 'firmware'. The main CPU on the Raspberry monitors
> # voltage drops and then acts on accordingly. If heavy voltage
> # drops directly after startup are monitored the firmware lowers
> # the core voltage available to CPU cores and also caps CPU
> # clockspeed. If voltage drops aren't that severe a more flexible
> # approach is used and at least you gain performance back after
> # periods of heavy load.

openoms, 2019-03-31:
I think the correct way to do this is from 2 terminal windows.
One running: `sysbench --test=cpu --cpu-max-prime=10000 --num-threads=4 run` ( install it with ` sudo apt install -y sysbench)
The other terminal running the `sudo ./test.sh` (1st download from the gist)

This way I had (sysbench running and stopping):
```
61.8'C 1400/1200 MHz 1.2313V
62.3'C 1400/1200 MHz 1.2313V
62.8'C 1400/1200 MHz 1.2313V
60.7'C  600/ 600 MHz 1.2V
58.5'C  600/ 600 MHz 1.2V
```
Taking the LCD off did not make a difference.

fluidvoice, 2019-03-31:
A script automating that test would be good to include in the initial install/config phase of Raspiblitz to alert the user up-front.  Perhaps we should also keep a doc tallying of all the PS's to stay away from and of course include the known good ones in the shopping lists.

fluidvoice, 2019-03-31:
Power supply: 2.5A Rhino model PSA-CA006  (basically it's garbage/unusable)

fluidvoice, 2019-03-31:
> I think the correct way to do this is from 2 terminal windows.
>
> One running: `sysbench --test=cpu --cpu-max-prime=10000 --num-threads=4 run` ( install it with ` sudo apt install -y sysbench)
>
> The [...]

Multiple runs with my "better" PS on my Pi3 (which has no heat sinks), no LCD, USB FD, USB HDD got up to:  76.3'C 1400/1200 MHz 1.2375V

openoms, 2019-03-31:
I have put it into a repo, so can be modified collaboratively and incorporated into the setup process: https://github.com/openoms/raspiblitz/blob/powertest/home.admin/XXpowertest.sh
sysbench is integrated, running for ~60 secs.
Will need to:
* stop the script automatically
* filter the output to show more readable messages about the PSU quality

To download and run: 
`wget https://raw.githubusercontent.com/openoms/raspiblitz/powertest/home.admin/XXpowertest.sh && sudo chmod +x XXpowertest.sh &&  ./XXpowertest.sh`
Once downloaded just run: ./XXpowertest.sh

fluidvoice, 2019-03-31:
Now we just need someone with the "official" Raspberry Pi 2.5A supply to run it, ideally, on a 3B+ with external USB HDD

openoms, 2019-03-31:
@fluidvoice Can you link to this PSU you mention as official? The shopping lists has been updated I think for 1.0 to recommend 3A supplies.

fluidvoice, 2019-03-31:
I read that the Pi 3B+ throttles around 60C degrees, and our tests show similar; throttled from 58C-63C.
So I wonder if the trottling is due more to the temperature than the voltage?

fluidvoice, 2019-03-31:
> @fluidvoice Can you link to this PSU you mention as official? The shopping lists has been updated I think for 1.0 to recommend 3A supplies.

official RPi-3 2.5Amp power supply (part number T5875DV): https://www.newark.com/raspberry-pi/t5875dv/psu-raspberry-pi-5v-2-5a-multi/dp/77Y6535

fluidvoice, 2019-03-31:
[image]

fluidvoice, 2019-03-31:
This guy is seeing throttling at 1.3438V volts even below 50C degrees: https://github.com/rootzoll/raspiblitz/issues/442#issuecomment-478367204

openoms, 2019-03-31:
> This guy is seeing throttling at 1.3438V volts even below 50C degrees: #442 (comment)

There is no throttling until the sysbench benchmarking is stopped after ~60 secs. There is 5 secs sleep between the measurements. So only relevant for 12 lines.

fluidvoice, 2019-03-31:
> > This guy is seeing throttling at 1.3438V volts even below 50C degrees: #442 (comment)
> 
> There is no throttling on that output. The sysbench benchmarking is stopped after ~60 secs. There is 5 secs sleep between the measurements, so the output is only relevant for 12 lines.

Interesting. His output is odd.   I'm wondering if the single line throttles are not really valid?  Because the OP throttles due to voltage or heat were more consistent, showing 3 to 5 lines at least of consistent throttling.  The spurious single-line... I wonder if it's just bad voltage regulation in the PS?

openoms, 2019-03-31:
The load is fluctuating with multiple services running in the background. See your output is very similar here: https://github.com/rootzoll/raspiblitz/issues/474#issuecomment-478340132 .
Could play with the the length of the load test and the measurement interval by editing the script or commands, but I doubt that it will be conclusive.

fluidvoice, 2019-03-31:
> The load is fluctuating with multiple services running in the background. See your output is very similar here: [#474 (comment)](https://github.com/rootzoll/raspiblitz/issues/474#issuecomment-478340132) .
> Could play with the the length of the load test and the measurement interval by editing the script or commands, but I doubt that it will be conclusive.

yes. good point/observation.  I was just biased towards this guy having a bad PS because of his many problems and file corruption that I don't know what's causing them.

openoms, 2019-03-31:
I have this: https://geekworm.com/collections/best-sellings/products/raspberry-pi-x820-2-5-inch-sata-storage-board-dc-5v-4a-power-supply-metal-case-kit with the 4A power supply. Powering with the small white micro-usb, because the LCD takes my GPIO. Throttling badly.
```
54.8'C 1400/ 600 MHz 1010000000000000101 1.2V
56.4'C 1400/ 600 MHz 1010000000000000101 1.2V
56.4'C 1400/ 600 MHz 1010000000000000101 1.2V
56.9'C 1400/ 600 MHz 1010000000000000101 1.2V
56.9'C 1400/ 600 MHz 1010000000000000101 1.2V
58.0'C 1400/ 600 MHz 1010000000000000101 1.2V
57.5'C 1400/ 600 MHz 1010000000000000101 1.2V
58.0'C 1400/ 600 MHz 1010000000000000101 1.2V
58.0'C 1400/ 600 MHz 1010000000000000101 1.2V
58.0'C 1400/ 600 MHz 1010000000000000101 1.2V
58.0'C 1400/ 600 MHz 1010000000000000101 1.2V
58.0'C 1400/ 600 MHz 1010000000000000101 1.2V
58.0'C 1400/ 600 MHz 1010000000000000101 1.2V
58.0'C 1400/ 600 MHz 1010000000000000101 1.2V
Done.

Maximum prime number checked in CPU test: 10000

Test execution summary:
    total time:                          71.7011s
    total number of events:              10000
    total time taken by event execution: 286.7295
    per-request statistics:
         min:                                 27.79ms
         avg:                                 28.67ms
         max:                                 66.48ms
         approx.  95 percentile:              32.71ms

Threads fairness:
    events (avg/stddev):           2500.0000/15.65
    execution time (avg/stddev):   71.6824/0.01

56.9'C 1400/ 600 MHz 1010000000000000101 1.2V
55.8'C  600/ 600 MHz 1010000000000000101 1.2V
56.4'C  600/ 600 MHz 1010000000000000101 1.2V
```
This is a great troubleshooting tool @fluidvoice. It could be the part of XXdebuglogs.sh maybe.

openoms, 2019-04-01:
> I have this: https://geekworm.com/collections/best-sellings/products/raspberry-pi-x820-2-5-inch-sata-storage-board-dc-5v-4a-power-supply-metal-case-kit with the 4A power supply. Powering with the small white micro-usb, because the LCD takes my GPIO. Throttling badly.

Same adapter board and PSU, but now powered through the GPIO port as shown here  (takes the place of the screen unfortunately):
[image]
 (takes the place of the screen unfortunately): 

```
34.9'C 1400 MHz 0000000000 1.3250V
40.8'C 1400 MHz 0000000000 1.3250V
42.9'C 1400 MHz 0000000000 1.3250V
44.0'C 1400 MHz 0000000000 1.3250V
46.2'C 1400 MHz 0000000000 1.3250V
47.2'C 1400 MHz 0000000000 1.3250V
Done.

Maximum prime number checked in CPU test: 10000

Test execution summary:
    total time:                          31.2306s
    total number of events:              10000
    total time taken by event execution: 124.8873
    per-request statistics:
         min:                                 11.90ms
         avg:                                 12.49ms
         max:                                 63.93ms
         approx.  95 percentile:              14.68ms

Threads fairness:
    events (avg/stddev):           2500.0000/28.38
    execution time (avg/stddev):   31.2218/0.00

42.9'C 1400 MHz 0000000000 1.3250V
41.9'C 1400 MHz 0000000000 1.3250V
40.8'C 1400 MHz 0000000000 1.3250V
41.3'C 1400 MHz 0000000000 1.3250V
40.8'C 1400 MHz 0000000000 1.3250V
41.9'C 1400 MHz 0000000000 1.3250V
41.9'C 1400 MHz 0000000000 1.3250V
42.4'C 1400 MHz 0000000000 1.3250V
42.4'C  600/ 600 MHz 0000000000 1.3250V
```
Max voltage achieved, no throttling and more than double the performance. (Load test calculating 1000 primes run in 31 secs vs 71 when there is undervoltage).

Also found evidence for throttling at 60 degrees:
```
52.1'C 1400 MHz  1.3250V
58.0'C 1400 MHz  1.3250V
59.6'C 1400 MHz  1.2313V
60.1'C 1400 MHz  1.2313V
60.7'C 1200/1400 MHz 1.2313V
60.7'C 1200/1400 MHz 1.2313V
59.1'C 1400 MHz  1.3250V
59.1'C 1400 MHz  1.2313V
59.6'C 1400 MHz  1.3250V
58.5'C 1400 MHz  1.3250V
59.1'C 1400 MHz  1.3250V
```
The XXpowertest.sh is updated. Now shows "RaspiBlitz powertest v0.1". Load stops after 60 seconds and the output after 15 lines (15x5sec = 75 secs for the whole thing)

fluidvoice, 2019-04-01:
During re-sync of the blockchain DB (which is annoyingly slow btw), the throttling is surprisingly low despite the high temperate.  My Pi is in the CryptoCloaks Lightning shell with heatsink on the CPU.

$ sudo ./test.sh 
To stop simply press [ctrl]-[c]

72.0'C 1400/1200 MHz 1.2563V
74.7'C 1400/1200 MHz 1.2563V
73.6'C  600/ 600 MHz 1.2V
73.6'C 1400/1200 MHz 1.2563V
74.1'C 1400/1200 MHz 1.2563V
73.6'C 1400/1200 MHz 1.2563V
74.1'C 1400/1200 MHz 1.2563V
73.6'C 1400/1200 MHz 1.2563V
75.2'C 1400/1200 MHz 1.2563V
75.8'C 1400/1200 MHz 1.2563V
75.8'C 1400/1200 MHz 1.2563V
76.3'C 1400/1200 MHz 1.2563V
75.2'C 1400/1200 MHz 1.2563V
74.1'C 1400/1200 MHz 1.2563V
75.2'C 1400/1200 MHz 1.2563V
76.3'C 1400/1200 MHz 1.2563V
75.2'C 1400/1200 MHz 1.2563V
74.7'C 1400/1200 MHz 1.2563V
^C

fluidvoice, 2019-04-01:
Based upon the warning Dietpi gives, I suppose I should be using a fan.  Something else the most users don't know about.

[image]

fluidvoice, 2019-04-01:
> Max voltage achieved, no throttling and more than double the performance. (Load test calculating 1000 primes run in 31 secs vs 71 when there is undervoltage).
>
> Also found evidence for throttling at [...]

NIce. I need a better power supply like that wall wart you are using.
Ideally the XXpowertest.sh should bang on the drive I/O at the same time as CPU load, say with ~150MB file R/W to simulate blockchain syncing.  Dunno if there is more power load pulled by the HDD when in use versus just spinning.

fluidvoice, 2019-04-01:
> Same adapter board and PSU, but now powered through the GPIO port as shown here (takes the place of the screen unfortunately):

Maybe connect that power lead directly to the Pi's micro-usb ...
http://wp.brodzinski.net/raspberry-pi-3b/custom-microusb-power-cord-for-raspberry-pi-3/

openoms, 2019-04-01:
Yes, we can play around with sysbench more (http://imysql.com/wp-content/uploads/2014/10/sysbench-manual.pdf). Making the HDD busy did not make any difference to the powertest output with a quick test, but it does add a lot of time.

I have a barrel connector-to-microUSB adapter already, so will test that.
I wonder if it would worth to start a google sheet to record the results for specific power supplies.

I expect a difference between powering the Pi through the GPIO and microUSB, it is not the same circuit.
The GPIO is more direct and therefore also theoretically more exposed to voltage spikes. But yes, defaulting to an HDMI screen would bring a lot of problems with cases.

openoms, 2019-04-01:
> But yes, defaulting to an HDMI screen would bring a lot of problems with cases
what do you mean here?

Just stating the obvious that an HDMI screen is not a good default option for the Pi mainly because the cases dont accomodate it will and it comes with more cables/connectors.  I have tested a HDMI screen with the Odroid XU4 successfully, however a good case is lacking there as well.

fluidvoice, 2019-04-01:
> > But yes, defaulting to an HDMI screen would bring a lot of problems with cases
> > what do you mean here?
> 
> Just stating the obvious that an HDMI screen is not a good default option for the Pi mainly because the cases dont accomodate it well and it comes with more cables/connectors. I have tested a HDMI screen with the Odroid XU4 successfully, however a good case is lacking there as well.

I wonder if the LCD would fit on top of this active cooling. Then maybe Cryptocloaks can 3D print a case for it.  https://geekworm.com/blogs/news/raspberry-pi-armor-case-aluminum-aolly-metal-enclosure-to-dress-up-your-raspberry-pi
[image]

openoms, 2019-04-01:
> I wonder if the LCD would fit on top of this heatsink w/fan cooling. Then maybe Cryptocloaks can 3D print a case for it. Same problem with the GPIO power pins though.
https://geekworm.com/blogs/news/raspberry-pi-armor-case-aluminum-aolly-metal-enclosure-to-dress-up-your-raspberry-pi

The left one will work and you don`t even need the a case on top it, because it is a case itself + you don`t want to impair the airflow. The connector of the fan would make take the place of the GPIO screen.

From the powertests so far we have seen that thermal throttling is not a huge problem (going from 1400 MHz to 1200), but a poor PSU is catastrophic being the the possible cause of HDD corruption and dropping the clock to 600 MHz.

fluidvoice, 2019-04-01:
> > I wonder if the LCD would fit on top of this heatsink w/fan cooling. Then maybe Cryptocloaks can 3D print a case for it. Same problem with the GPIO power pins though.
> > https://geekworm.com/blogs/news/raspberry-pi-armor-case-aluminum-aolly-metal-enclosure-to-dress-up-your-raspberry-pi
> 
> The left one will work and you don`t even need the a case on top it, because it is a case itself + you don`t want to impair the airflow. The connector of the fan would take the place of the GPIO screen, so not an option.
> 
> From the powertests so far we have seen that thermal throttling is not a huge problem (going from 1400 MHz to 1200), but a poor PSU is catastrophic being the the possible cause of HDD corruption and dropping the clock to 600 MHz.

The heatsink without a fan will do very little when demand is long duration (like during IBD - initial blockchain download, or re-syncing).  I've seen tests done on passive vs active cooling.  A fan is needed for any significant change in temps.

openoms, 2019-04-01:
> To be honest, 3.5A is really pushing the limits of the microUSB connector. That's your bottleneck, as the contacts are so small they add appreciable resistance, which lowers your voltage as the current increases.

Good point. No wonder why all the other SBC-s I know have a barrel connector.

ThomasKaiser, 2019-04-01:
https://forum.openmediavault.org/index.php/Thread/18991-New-approach-for-Raspberry-Pi-OMV-images/?postID=190184#post190184

openoms, 2019-04-01:
> https://forum.openmediavault.org/index.php/Thread/18991-New-approach-for-Raspberry-Pi-OMV-images/?postID=190180#post190180

@ThomasKaiser Thank you for pointing to this.
In there you write: 
>So the new behavior is documented at least there. To fix things occurrences of printf "%019d" should be replaced with printf "%021d"

I have tested that, but still get the `Numerical result out of range` error.
What should happen with the 
`Health=$(perl -e "printf \"%19b\n\", $(sudo vcgencmd get_throttled | cut -f2 -d=)")` ?
In place of the `%19b` I have tried `%21b` but getting this:
```
./XXpowertest.sh: line 32: printf: warning: 10000000000000000000: Numerical result out of range
55.3'C 1400 MHz 009223372036854775807 1.3250V
./XXpowertest.sh: line 35: printf: warning: 10000000000000001000: Numerical result out of range
60.1'C 1400/1200 MHz 009223372036854775807 1.3250V
```

with %021b the error is gone, but the number still does not make sense:
```
66.6'C 1400/1200 MHz 000144115188075856384 1.2313V
67.1'C 1400/1200 MHz 000144115188075856384 1.2313V
```

Sorry I don`t know much about the binary conversion.

ThomasKaiser, 2019-04-01:
Maybe RPi Trading employees changed stuff in ThreadX again and you spot it somewhere [here](https://github.com/raspberrypi/firmware/commits/master)? I mean, it's a closed source world you're dealing here with so good luck. Can't test myself since not using Raspberries for obvious reasons (way too lousy for the use cases I'm interested in).

You might ask over at https://github.com/raspberrypi/firmware/commit/404dfef3b364b4533f70659eafdcefa3b68cd7ae#commitcomment-31620514 (some of the guys are nice and helpful).

openoms, 2019-04-01:
Right, thanks for the insight. I am not a fan of the closed source either, but did not really come across any truly open source hardware yet. The Odroids and Pine64 meant to be but I`ve reached a closed floor there as well, the difference is only in how deep one can go.

fluidvoice, 2019-04-02:
> > To be honest, 3.5A is really pushing the limits of the microUSB connector. That's your bottleneck, as the contacts are so small they add appreciable resistance, which lowers your voltage as the current increases.
> 
> Good point. No wonder why all the other SBC-s I know have a barrel connector.

It's "better" for careful people.  But I can see why they didn't make it default on the Pi.  There are too many diff barrel connector sizes, supply voltages, etc.

rootzoll, 2019-04-02:
@fluidvoice @openoms great reasearch 👍 I tried to get up to speed. So here are mare short term takeaways for the next release 1.2 so far - please feel free to comment:

- update shopping lists with better power supplies & cables
- add heat sinks to shopping lists (as long as we dont have a fan solution)
- on fresh sd card (setup/update) run a longer testphase to check power situation
- on XXdebugLogs.sh add a short power analysis (outage count and threadX voltages)

fluidvoice, 2019-04-02:
> @fluidvoice @openoms great reasearch  I tried to get up to speed. So here are mare short term takeaways for the next release 1.2 so far - please feel free to comment:
> 
> * update shopping lists with better power supplies & cables
> * add heat sinks to shopping lists (as long as we dont have a fan solution)
> * on fresh sd card (setup/update) run a longer testphase with load to check power situation
> * on XXdebugLogs.sh add a short power analysis (outage count and threadX voltages)

* need to find and add cheap fan solutions. As said above, heatsink only does almost nothing for long duration loads, such as when blockchain syncing.  In time perhaps we can find decent case-with-fan combo's.

ThomasKaiser, 2019-04-02:
> run a longer testphase with load to check power situation

Pretty easy and done in seconds by using [cpuburn](https://github.com/ssvb/cpuburn-arm). With unreliable powering firing up `cpuburn-a53` on the majority of RPi 3 will result in the board powering off immediately. So it's a multi step procedure:

* limit `cpufreq_max` to 600 MHz, fire up `cpuburn-a53` (or `cpuburn-a7` on original RPi 2) for 2 seconds and check ThreadX voltage. If below a certain value stop here and report 'powering insufficient'
* limit `cpufreq_max` to 1200 MHz and test again 2 seconds. If voltage below a certain value stop here
* unlock `cpufreq_max` to 1400 MHz and test again

> heatsink only does almost nothing for long duration loads

You know that you can set `temp_soft_limit=70` in `/boot/config.txt` to revert back to previous behavior before RPi Trading guys decided to [trash RPi 3B+ performance on the entire planet just to masquerade some QA problems they had with the first production batch](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=217056)?

Maybe the RPi 3B+ footnotes in [my sbc-bench results overview](https://github.com/ThomasKaiser/sbc-bench/blob/master/Results.md) are also of interest.

fluidvoice, 2019-04-02:
> Maybe the RPi 3B+ footnotes in [my sbc-bench results overview](https://github.com/ThomasKaiser/sbc-bench/blob/master/Results.md) are also of interest.

yes! very interesting indeed. Thanks for compiling that great list.

ThomasKaiser, 2019-04-02:
> DietPi runs OK on the NEO4 board?

Dietpi is not a Debian based distro (being bootstrapped from scratch). It's just the ignorant attempt to take someone else's Debian OS image and then let a script modify the userspace obsessed by the idea the less packages would be installed the faster the resulting system would be (which is BS -- all the relevant stuff happens somewhere else instead in this century). And then they clean up the mess somewhat manually and provide this hand-crafted 'new' image rebranded as "DietPi for ...".

They take what they find somewhere on the net and then stumble across 'issues' only by accident, see [this nice example](https://github.com/MichaIng/DietPi/issues/1654) of a cronjob active to improve NAS performance. Could've been also a backdoor. Or just a simple daemon stealing Wi-Fi logon credentials. You never know. DietPi images are not created from scratch so you are **not** in the position to trust into upstream projects (like Linux kernel and Debian) or audit stuff. So your only chance is to **believe** that the DietPi image you downloaded is 100% error/vulnerability/backdoor free which is a funny assumption given the way DietPi images are created (manually) and based on (random stuff from the Internet).

If the base image they took to transform into "DietPi for ..." was set up in a way that a separate repository for bootloader and kernel updates is configured (none of the images they base on is using the ARM kernels provided by upstream Debian), then you will get those security fixes later on 'with DietPi', otherwise not. With DietPi it's possible running 'an OS' that is cut off from important security fixes and DietPi folks don't care since they don't care about such 'low level' stuff like bootloader/kernel and security in general.

No idea whether FriendlyELEC OS images they base their 'DietPi for NanoPi ...' stuff on provides a kernel repo or not. I never use vendor provided OS images on any SBC since I don't feel comfortable having to trust in some human beings somewhere manually fiddling around in an OS image trying to do the right things. I create my OS images from scratch using [an appropriate build system](https://github.com/armbian/build) so I only have to trust in upstream Debian/Ubuntu and Linux developers and not random human beings manually modifying stuff they often barely understand :)

fluidvoice, 2019-04-02:
@ThomasKaiser thanks for the input. It's def something to think about esp since this project involves running a crypto-currency node and wallets.  Perhaps we will experiment/test building on top of Debian base. Do you feel similar to above about Armbian?

openoms, 2019-04-02:
@ThomasKaiser thank you for this extensive explanation. I understand your concerns now.
I just got an Odroid XU4 with a Cloudshell2  RAID board (https://ameridroid.com/collections/cases/products/cloudshell2-nas-kit) so might as well test Armbian on it. 
I like that there are at least one (2 most in most cases) less teams tinkering with the original resource there.

openoms, 2019-04-02:
This whole changing platforms question is still in the research phase so I linked your comment to the relating topic. As I understand according to the presented logic Armbian would be even superior to Raspbian in terms of security (and maybe performance)?

ThomasKaiser, 2019-04-02:
> drawn to DietPi due to the user-base, support, knowledge, workarounds, etc., with various SBC's

They only deal with the userspace and completely ignore all the relevant bits (bootloader, kernel, low-level settings, **security** in general). "DietPi for..." can be a pile of crap when the original image they chose was crap while when the original image was a good one taking care of kernel updates, a recent kernel version and appropriate settings, there are a lot less concerns wrt the resulting "DietPi" (which is just a stripped down userland, some scripts and menu-driven tools aiding inexperienced users on top of the relevant **base system** they took from somewhere else).

DietPi on Raspberries greatly benefits from all relevant settings as well as bootloader/kernel updates provided by RPi Trading Ltd. folks being part of the base image. On any other board you need to check twice since it's hit-and-miss there. Can be an image running an outdated kernel 3.4 which didn't receive any fixes since ages using brain-dead so called DVFS settings leading to an overheating board even in idle. Can also be a base image that does most of the stuff correctly and even includes a repository to fetch kernel updates from (so you get updates but need to trust into that repo or the people behind! Delivering an evil backdoor through `apt upgrade` is really easy once you control the repo).

And that's IMO the main problem with Linux on ARM: security/trust as long as the device in question is not fully [SBSA](https://en.wikipedia.org/wiki/Server_Base_System_Architecture) compliant or supports at least GRUB-ARM-EFI to allow to run upstream distros like Debian/RedHat/SuSE without the need to use special OS images.

Same applies to Armbian (I know this project quite well since being a former developer) but at least here you can generate whole OS images 100% from scratch on your own machine without having to trust in some individuals knowing what they do. You 'only' need to trust into upstream Debian/Ubuntu and Linux kernel teams doing their job. The remaining problem is where you get the kernel updates from. If it's apt.armbian.com you obviously need to trust the owner of this host now. Upstream projects like Debian have several maintainers/contributors and working peer review. With small projects there's often nothing. But with an Armbian generated image it's easy to pull in such kernel updates from somewhere else (your own repository if you have the knowledge to build kernels and package them accordingly).

TL;DR: there's no easy solution especially if you need to take care about security :)

ThomasKaiser, 2019-04-02:
> No idea whether FriendlyELEC OS images they base their 'DietPi for NanoPi ...' stuff on provides a kernel repo or not

Even if FriendlyELEC provides such a repo DietPi isn't using it: https://github.com/MichaIng/DietPi/issues/1829#issuecomment-429256961 (which means they have no mechanism to update the kernel from within the image on this platform and the kernel they use has been provided by $someone who put files on WeTransfer)

> As I understand according to the presented logic Armbian would be even superior to Raspbian in terms of security (and maybe performance)?

Armbian doesn't support Raspberries directly though I prepared OMV images for Raspberry Pi 2 and 3 created by Armbian's build system using [Armbian's customization feature](https://github.com/ThomasKaiser/OMV4_for_Raspberries) and then manually combining the Armbian rootfs with an OS image skeleton suitable for RPi (the ThreadX stuff on the FAT partition being the most important part).

Wrt performance please see https://github.com/bavison/arm-mem/commit/ea0bde27d823bd27f2e38ea0913f124b540c5ecc#commitcomment-32793945 and https://www.cnx-software.com/2019/03/13/raspberry-pi-suddenly-not-working-repair-microsd-card/#comment-561518 (the TL;DR version: Raspbian is performance optimized for some specific use cases like using Chromium while for most other use cases a plain Debian armhf userland as Armbian provides performs slightly better).

Wrt security the situation with RPi is a mess in general due to the closed source nature of the primary OS (ThreadX) but on the other hand RPi Trading employees are pretty good in providing security fixes rather timely. Take [Dirty COW](https://en.wikipedia.org/wiki/Dirty_COW) for example: RPi Trading guys provided a fix on Oct 25th (6 days after disclosure) while in Armbian we had patched all kernels and provided fixed kernel packages for all +50 boards [already on Oct 22th](https://github.com/armbian/build/search?q=dirtycow&type=Commits). 

With DietPi everything did depend on the base image in question DietPi folks to choose to 'transform' into DietPi. DietPi for RPi as such got the fix on Oct 25th due to archive.raspberrypi.org repo configured for kernel updates, other DietPi images back then were based on Armbian and as such received the fixes days earlier, other DietPi images based on crappy base images never received any fix.

openoms, 2019-04-03:
> Armbian doesn't support Raspberries directly though I prepared OMV images for Raspberry Pi 2 and 3 created by Armbian's build system using Armbian's customization feature and then manually combining [...]

This project is based on the RaspberryPi, but I get implications with the closed source ThreadX. By default we will probably need to rely on Raspbian, but that does not stop anyone to use other images for different SBC-s. Some need minimal change in our script (like the Odroid HC1 / XU4) , some significantly more depending on their processor architecture and the base image. DietPi was supposed to provide a fairly standard platform, but I see it is far from perfect and also varies between boards. 

> The NanoPi NEO4 looks like a good inexpensive candidate

@fluidvoice This Neo4 board is ok, but it needs the 64bit aarch64 builds of bitcoind and lnd vs the armv6 (32 bit) versions what the RPi uses.
I have already set up the aarch64 options for Rock64, so possibly can use that branch to test this board. It is all about someone testing it out.

ThomasKaiser, 2019-04-03:
> DietPi was supposed to provide a fairly standard platform, but I see it is far from perfect and also varies between boards

DietPi only cares about the userland. And it does not vary 'between boards' but 'between base images' instead.

Take RK3399 as an example:

* On the FriendlyELEC RK3399 devices they deal with a bizarre Android image format even needing [some obscure Windows tools to unpack/repack the whole image](https://github.com/MichaIng/DietPi/issues/1829#issuecomment-397845348). Why? Since FriendlyELEC built their base image in this way and DietPi doesn't deal with 'low level stuff' (as they call it).
* 'DietPi for RockPro64' instead uses a GPT partition layout with 7 partitions. Why? Since @ayufan decided to follow the Rockchip recommendation for Linux
* 'DietPi for ODROID N1' uses again a totally different partition layout. Why? Since Hardkernel and @meveric decided to use something else

With DietPi and with one of these devices situation wrt kernel patches is completely different but it's again not about 'boards' but the base image they found somewhere on the net to base on. With RockPro64 you receive kernel images since @ayufan takes care, with FriendlyELEC devices no kernel updates at all and with ODROID-N1 while the infrastructure provided by @meveric is there most probably no updates will be provided since N1 has been canceled last year (I only added it to the comparison as a 3rd example for a totally different partition layout used with the same SoC)

Boards of the same board family should be treated identical since everything else doesn't make that much sense. But with DietPi it always and only depends on which OS image they found somewhere on the net to base their userland modifications on (to provide the handcrafted result then rebranded as 'DietPi for ...')

Even worse since they only care about the userland everything really relevant is considered out of their scope and will be treated as 'External Bug'. Unfortunately DietPi is really just a bunch of scripts to modify the userland of an otherwise already existing Debian OS image. Great for inexperienced users needing menus even for most simply stuff but not that great if you got the impression it would be an own distro supporting a variety of different hardware (especially in a consistent way)

openoms, 2019-04-03:
@ThomasKaiser thank you again for taking the time to explain all this, you clearly have a lot of insight about various SBC-s and will continue to follow you work.

My conclusion is:
* so far we got a proof of concept of the RaspiBlitz system working reliably on other SBC-s and can be compatible with updates.
*  it is far from being possible to provide a platform which can run easily on different boards without serious security trade-offs
* if we are testing other SBC-s a best base distro should be picked individually for every board according to what is updated the most frequently and reliably.
* this is the limitation of Linux on Arm, but we will continue to work with that given the price and form factor is unmatched in Intel based systems.

ThomasKaiser, 2019-04-03:
The issue of trust is missing. Security updates are important but where do they come from? If you rely on a Debian userland these fixes are provided by Debian teams. On x86 also kernel fixes are provided by Debian and while the same is true for ARM in general none of the popular OS images for the more modern SBC out there uses Debian's kernel since way behind (only really old SBC are fully supported by Debian's ARM kernel and [obviously this kernel doesn't receive enough testing anyway](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=922478)).

Wrt kernel patches on ARM we end up looking at individual human beings and not organized teams with established peer review processes almost everywhere. And this could be an issue in its own if you seriously think about security.

BTW: situation with Raspbian is comparable. Raspbian is **not** real Debian since while they use upstream Debian package sources they have to build every package themselves often running into compatibility issues therefore delaying delivery of packages (same problem is the reason Raspbian [never implemented Debian's multiarch support](https://www.cnx-software.com/2016/03/01/64-bit-arm-aarch64-instructions-boost-performance-by-15-to-30-compared-to-32-bit-arm-aarch32-instructions/#comment-523286) to provide 32-bit and 64-bit binaries at the same time).

So it's not a Debian team providing packages (kernel and ThreadX included) but a few RPi Trading employees instead you need to trust.

openoms, 2019-04-03:
Yes, I understand this completely, using an SBC at all means we are taking a security compromise.

Please keep in mind that the RaspiBlitz is a project to make it the cheapest and easiest for people to go down the rabbit hole of running their own full Bitcoin and Lightning node, including total noobs. It is well emphasised in the readme that it is not for storing significant amount of value. Obviously the whole Lightning Network is under development and has this the #RECKLESS going for a reason.

For people more savvy and security focused there are plenty other alternatives:
https://github.com/seth586/guides/blob/master/FreeNAS/README.md
https://medium.com/lightning-power-users/lightning-routing-node-starter-pack-704c0e7d79cb

Saying this I see the RaspiBlitz system itself being quite security conscious and making it compatible with more reliable and performant bases (even x86) is a great side project I`d like to continue.

fluidvoice, 2019-04-03:
@openoms this heatsink looks like it might work in most cases even with the 3.5" LCD on top as it looks lower than the USB sockets and the LCD sits a little above it.  I'll prob get one of these to try out. They go from as low as $6.50 to $12 depending where you buy.  
video review of dual-fan heatsink: https://www.youtube.com/watch?v=j_QrsNwVwNA
https://www.amazon.com/iUniker-Raspberry-Dual-Fan-Heatsink/dp/B07D5WWNH6/ref=as_li_ss_tl
Another single-fan low profile:
https://www.banggood.com/3510-Version-Extreme-Cooling-Fan-Copper-Heatsink-Thermal-Tapes-Kit-For-Raspberry-Pi-3B-p-1411599.html

openoms, 2019-04-03:
> Obviously one must apply a workaround for powering the fans.

Looks cool and probably efficient, but would like to see that workaround

fluidvoice, 2019-04-03:
> > Obviously one must apply a workaround for powering the fans.
> 
> Looks cool and probably efficient, but would like to see that workaround

I have some ideas but will hold off until I've been able to try them.  It's a shame there are not low-profile GPIO pin "taps" that I can find. Ie, thin plate-like contacts that don't raise the LCD up.

[26 more replies]

closed 2019-04-11 by rootzoll