raspiblitz/raspiblitz #478: LCD screen not working properly with V1.1 (shows the CLI but not the Blitz GUI)

Z-and, 2019-03-31:
I'm in the process of setting up the RaspiBlitz (have previously setup the RaspiBolt) and the first thing I'm noticing is the LCD screen doesn't seem to match what's displayed on PuTTY. When I fire up the Pi the LCD screen will shoot a bunch of lines in the CLI (hard to read because of how small the screen is) and then just kinda gets stuck at one particular line. When I connect via PuTTY, the normal Blitz GUI comes up and I'm able to go through the configuration process but the LCD screen doesn't change at all. It stays stuck on the CLI with the same code that initialized during startup. Any ideas?

fluidvoice, 2019-03-31:
Post what the LCD shows and also what shows on the HDMI monitor.

Z-and, 2019-03-31:
I've attached a couple photos. One of the LCD screen (hard to really capture), and one of my PuTTY screen (I'm currently syncing the blockchain). Also below is a link to the exact screen I bought off Amazon, looks to be the same one one the shopping list

https://www.amazon.com/gp/product/B075JFHLQD/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
[image]
[image]

openoms, 2019-03-31:
This is an HDMI screen, not the same to the ones in the shopping list.

openoms, 2019-03-31:
Run this:
```
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./MPI3508-show

    After executing the above command, the system will restart automatically, and display and touch will be OK.```

Source: http://www.lcdwiki.com/3.5inch_HDMI_Display-B

If the letters will be still too small, you will need to resize the font.

Z-and, 2019-03-31:
> This is an HDMI screen, not the same to the ones in the shopping list.

I see, will it still be compatible with this setup? There is an HDMI to HDMI part that connects the Pi to the screen, should I remove that and only connect with the pins?

fluidvoice, 2019-03-31:
> > This is an HDMI screen, not the same to the ones in the shopping list.
> 
> I see, will it still be compatible with this setup? There is an HDMI to HDMI part that connects the Pi to the screen, should I remove that and only connect with the pins?

There is pins?  I thought this is HDMI connection only?  Don't remove the HDMI connector part.

Z-and, 2019-03-31:
> There is pins? I thought this is HDMI connection only? Don't remove the HDMI connector part.

Yeah there are both pins and an HDMI connector.

fluidvoice, 2019-03-31:
picture please

Z-and, 2019-03-31:
It looks exactly like this except there is also an HDMI port, an Ext port, and what seems to be a headphone jack. I would take a photo of mine but I'm afraid to remove it since I'm in the middle of a sync.

[image]

Z-and, 2019-03-31:
> picture please

It looks exactly like this except there is also an HDMI port, an Ext port, and what seems to be a headphone jack. I would take a photo of mine but I'm afraid to remove it since I'm in the middle of a sync.

[image]

openoms, 2019-03-31:
@Z-and It probably looks like this: [image]
@fluidvoice you can see the HDMI connector on his pic: 
[image]

It will be compatible and you can connect it to the pins, but it only gets the power from there.
It can be also powered through the microUSB so you can have it separated from the Pi.
Once your sync is finished, just install it as described: https://github.com/rootzoll/raspiblitz/issues/478#issuecomment-478385232

Z-and, 2019-04-01:
> @Z-and It probably looks like this:
> [image]
> @fluidvoice you can see the HDMI connector on his pic:
> [image]
> 
> It will be compatible and you can connect it to the pins, but it only gets the power from there.
> It can be also powered through the microUSB so you can have it separated from the Pi.
> Once your sync is finished, just install it as described: [#478 (comment)](https://github.com/rootzoll/raspiblitz/issues/478#issuecomment-478385232)

Thank you, that worked! Now I just need to figure out why my sync keeps failing. I setup Bitcoin Core 0.17.1 config file with txindex=1, and transfered all the necessary folders (blocks, chainstate, indexes) but it keeps giving me the following message. 

[image]

openoms, 2019-04-01:
I see you opened a new issue for the sync problems, so you can close this one. (I can`t)

Basicly you need to make sure that you have restarted your bitcoin client on the windows with txindex=1 and also closed it before starting the scp. https://github.com/rootzoll/raspiblitz/issues/479#issuecomment-478446366

Z-and, 2019-04-01:
Resolved #478

Z-and, 2019-04-01:
> Run this:
> 
> ```
> git clone https://github.com/goodtft/LCD-show.git
> chmod -R 755 LCD-show
> cd LCD-show/
> sudo ./MPI3508-show
> ```
> After executing the above command, the system will restart automatically, and display and touch will be OK.
> 
> Source: http://www.lcdwiki.com/3.5inch_HDMI_Display-B
> 
> If the letters will be still too small, you will need to resize the font which is described in this comment: [#244 (comment)](https://github.com/rootzoll/raspiblitz/issues/244#issuecomment-476713706)

Fixes #478

Z-and, 2019-04-01:
> Run this:
>
> ```
> git clone https://github.com/goodtft/LCD-show.git
> chmod -R 755 LCD-show
> cd LCD-show/
> sudo ./MPI3508-show
> ```
> After executing the above command, the system will restart automatically, [...]

Fixes #478

closed 2019-04-01 by Z-and