raspiblitz/raspiblitz #482: Price indicator on display

WetPuppyKisses, 2019-04-02:
This may be not strictly an issue but I would like some help to add the Bitcoin price in dollars on the display of the RaspiBlitz (Once fully synced)

I am not expert on bash programming, but I took as reference the following:

https://github.com/84adam/wink.sh/blob/master/wink.sh
https://github.com/8go/coinbash

The ideal would be that the price is automatically updated every 15 minutes or so.

rootzoll, 2019-04-02:
In the mid term we will change from the LCD ASCII mode to the LCD graphics mode with maybe even more info. This could be part of that.

nyxnor, 2021-04-18:
please close this. @raulcano started this, but now maybe the lcd wont work. Also, webui will solve this much easier.