raspiblitz/raspiblitz #483: ver. 1.1

vamp111, 2019-04-02:
after updating to ver.1.1 everyday my Raspi stops working normaly and gets stuck at "lightning filtering blockchain" ?/5******* screen

my RTL as well is unable to load any data. Here my log file, let me know if you need anything else

*** RASPIBLITZ LOGS ***
blitzversion: 1.1
chainnetwork: bitcoin / main
 09:34:36 up 12 min,  2 users,  load average: 4.64, 3.97, 2.29

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-04-02 09:22:44 BST; 11min ago
  Process: 834 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 849 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─849 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoi…id

Apr 02 09:22:42 FreeTradeBulgaria systemd[1]: Starting Bitcoin daemon...
Apr 02 09:22:44 FreeTradeBulgaria systemd[1]: Started Bitcoin daemon.

*** LAST 5 ERROR LOGS ***
-- Logs begin at Tue 2019-04-02 09:21:36 BST, end at Tue 2019-04-02 09:34:36 BST. --
Apr 02 09:22:42 FreeTradeBulgaria systemd[1]: Starting Bitcoin daemon...
Apr 02 09:22:44 FreeTradeBulgaria systemd[1]: Started Bitcoin daemon.

*** LAST 20 INFO LOGS ***
2019-04-02T08:24:28Z init message: Done loading
2019-04-02T08:24:28Z msghand thread start
2019-04-02T08:24:36Z txindex is enabled at height 569848
2019-04-02T08:24:36Z txindex thread exit
2019-04-02T08:24:39Z Loading addresses from DNS seeds (could take a while)
2019-04-02T08:24:40Z New outbound peer connected: version: 70015, blocks=569848, peer=0
2019-04-02T08:24:53Z New outbound peer connected: version: 70015, blocks=569848, peer=1
2019-04-02T08:24:54Z 239 addresses found from DNS seeds
2019-04-02T08:24:54Z dnsseed thread exit
2019-04-02T08:24:54Z New outbound peer connected: version: 70015, blocks=569848, peer=2
2019-04-02T08:24:55Z New outbound peer connected: version: 70015, blocks=569848, peer=3
2019-04-02T08:24:55Z New outbound peer connected: version: 70015, blocks=569848, peer=4
2019-04-02T08:24:56Z New outbound peer connected: version: 70015, blocks=569848, peer=5
2019-04-02T08:24:56Z New outbound peer connected: version: 70015, blocks=569848, peer=6
2019-04-02T08:24:57Z New outbound peer connected: version: 70015, blocks=569848, peer=7
2019-04-02T08:25:30Z New outbound peer connected: version: 70015, blocks=569848, peer=8
2019-04-02T08:27:44Z UpdateTip: new best=0000000000000000000bd70a5838ca0f2293f3ef4c50560c21afacfd84572312 height=569849 version=0x20000000 log2_work=90.502608 tx=397700040 date='2019-04-02T08:27:14Z' progress=1.000000 cache=1.6MiB(13792txo) warning='40 of last 100 blocks have unexpected version'
2019-04-02T08:29:23Z Pre-allocating up to position 0x1100000 in rev01581.dat
2019-04-02T08:29:24Z UpdateTip: new best=000000000000000000299eb48194888aa530770fb2a7ce8597ded078931a4730 height=569850 version=0x20000000 log2_work=90.502631 tx=397702493 date='2019-04-02T08:27:41Z' progress=0.999999 cache=2.3MiB(20827txo) warning='39 of last 100 blocks have unexpected version'
2019-04-02T08:31:09Z UpdateTip: new best=0000000000000000000478d9047e8abc561ef799406934a86bd141938fa084fc height=569851 version=0x20000000 log2_work=90.502654 tx=397705233 date='2019-04-02T08:29:26Z' progress=0.999999 cache=3.4MiB(29792txo) warning='39 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-04-02 09:30:00 BST; 4min 36s ago
 Main PID: 5796 (lnd)
   CGroup: /system.slice/lnd.service
           └─5796 /usr/local/bin/lnd --externalip=78.130.148.218

Apr 02 09:30:00 FreeTradeBulgaria systemd[1]: Started LND Lightning Daemon.

*** LAST 5 LND ERROR LOGS ***
-- Logs begin at Tue 2019-04-02 09:21:36 BST, end at Tue 2019-04-02 09:34:37 BST. --
Apr 02 09:29:00 FreeTradeBulgaria systemd[1]: lnd.service: Unit entered failed state.
Apr 02 09:29:00 FreeTradeBulgaria systemd[1]: lnd.service: Failed with result 'exit-code'.
Apr 02 09:30:00 FreeTradeBulgaria systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Apr 02 09:30:00 FreeTradeBulgaria systemd[1]: Stopped LND Lightning Daemon.
Apr 02 09:30:00 FreeTradeBulgaria systemd[1]: Started LND Lightning Daemon.

*** LAST 20 LND INFO LOGS ***
2019-04-02 09:32:13.127 [INF] DISC: Authenticated Gossiper is starting
2019-04-02 09:32:13.127 [INF] BRAR: Starting contract observer, watching for breaches.
2019-04-02 09:32:13.127 [INF] NTFN: New block epoch subscription
2019-04-02 09:32:13.144 [INF] CRTR: FilteredChainView starting
2019-04-02 09:32:17.561 [DBG] DISC: Skipping retransmission of channel without AuthProof: 624555589996969984
2019-04-02 09:32:27.082 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:32:27.190 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:32:28.275 [INF] RPCS: [listchannels] fetched 25 channels from DB
2019-04-02 09:32:36.784 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:32:36.911 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:32:37.979 [INF] RPCS: [listchannels] fetched 25 channels from DB
2019-04-02 09:32:43.032 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:32:43.157 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:32:44.229 [INF] RPCS: [listchannels] fetched 25 channels from DB
2019-04-02 09:33:03.564 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:33:03.667 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:33:04.697 [INF] RPCS: [listchannels] fetched 25 channels from DB
2019-04-02 09:33:37.075 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:33:37.191 [DBG] RPCS: [walletbalance] Total balance=0.00437857 BTC
2019-04-02 09:33:38.326 [INF] RPCS: [listchannels] fetched 25 channels from DB

*** LAST 20 RTL LOGS ***
-- Logs begin at Tue 2019-04-02 09:21:36 BST, end at Tue 2019-04-02 09:34:37 BST. --
Apr 02 09:22:44 FreeTradeBulgaria systemd[1]: Started RTL daemon.

vamp111, 2019-04-02:
updated:

*** RASPIBLITZ LOGS ***
blitzversion: 1.1
chainnetwork: bitcoin / main
 11:50:51 up  2:29,  2 users,  load average: 0.74, 0.90, 0.87

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-04-02 09:22:44 BST; 2h 28min ago
  Process: 834 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 849 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─849 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoi…id

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST 5 ERROR LOGS ***
-- No entries --

*** LAST 20 INFO LOGS ***
2019-04-02T08:52:18Z Imported mempool transactions from disk: 15003 succeeded, 3981 failed, 0 expired, 2 already there
2019-04-02T08:55:37Z UpdateTip: new best=000000000000000000008ecdc3d8a551c22ea5c7fe0c342d7fac187da895a9ff height=569857 version=0x20c00000 log2_work=90.502789 tx=397720030 date='2019-04-02T08:55:22Z' progress=1.000000 cache=11.0MiB(96939txo) warning='41 of last 100 blocks have unexpected version'
2019-04-02T09:13:22Z UpdateTip: new best=00000000000000000017520034dcc4081bb9d461ae451de21f2b872ae8348bee height=569858 version=0x20000000 log2_work=90.502811 tx=397722520 date='2019-04-02T09:13:11Z' progress=1.000000 cache=12.6MiB(108395txo) warning='41 of last 100 blocks have unexpected version'
2019-04-02T09:28:03Z UpdateTip: new best=00000000000000000006ea1899f07aee21f600172a928f1fab8a75a43f712d1a height=569859 version=0x20000000 log2_work=90.502834 tx=397724786 date='2019-04-02T09:27:32Z' progress=1.000000 cache=14.0MiB(121839txo) warning='41 of last 100 blocks have unexpected version'
2019-04-02T09:31:51Z Pre-allocating up to position 0x200000 in rev01582.dat
2019-04-02T09:31:51Z UpdateTip: new best=000000000000000000210f90433e99bb11a5af2516ad9281e67a71c5925cd4d7 height=569860 version=0x20400000 log2_work=90.502856 tx=397727605 date='2019-04-02T09:31:28Z' progress=1.000000 cache=14.9MiB(130189txo) warning='41 of last 100 blocks have unexpected version'
2019-04-02T09:32:19Z UpdateTip: new best=000000000000000000147cd80895cd94e5df7e9222cfcec17efab148a580acff height=569861 version=0x20000000 log2_work=90.502879 tx=397727606 date='2019-04-02T09:31:47Z' progress=1.000000 cache=14.9MiB(130319txo) warning='41 of last 100 blocks have unexpected version'
2019-04-02T09:33:23Z UpdateTip: new best=00000000000000000018c7b1eb50ff05e28f5ee2118e5b5838ac815a306893de height=569862 version=0x20400000 log2_work=90.502901 tx=397728418 date='2019-04-02T09:33:02Z' progress=1.000000 cache=15.0MiB(131171txo) warning='42 of last 100 blocks have unexpected version'
2019-04-02T09:36:29Z UpdateTip: new best=0000000000000000000776e21afb2632735f1453c4e495400d747d495eef1192 height=569863 version=0x20800000 log2_work=90.502924 tx=397730945 date='2019-04-02T09:35:56Z' progress=1.000000 cache=15.5MiB(135491txo) warning='43 of last 100 blocks have unexpected version'
2019-04-02T10:08:37Z Potential stale tip detected, will try using extra outbound peer (last tip update: 1928 seconds ago)
2019-04-02T10:08:37Z New outbound peer connected: version: 70015, blocks=569863, peer=18
2019-04-02T10:09:06Z New outbound peer connected: version: 70015, blocks=569863, peer=19
2019-04-02T10:10:25Z UpdateTip: new best=00000000000000000003aca514677299b999e8af5d23a3900f666a82c761a8b6 height=569864 version=0x20000000 log2_work=90.502946 tx=397733687 date='2019-04-02T10:09:38Z' progress=1.000000 cache=17.3MiB(151909txo) warning='43 of last 100 blocks have unexpected version'
2019-04-02T10:16:43Z UpdateTip: new best=000000000000000000060da071c540140a799ae8882011ab264275158e6863fc height=569865 version=0x20000000 log2_work=90.502969 tx=397736434 date='2019-04-02T10:16:45Z' progress=1.000000 cache=18.1MiB(160085txo) warning='43 of last 100 blocks have unexpected version'
2019-04-02T10:30:28Z UpdateTip: new best=000000000000000000232172c1c149101269e5579d66024c0980f51fc84f98d3 height=569866 version=0x20000000 log2_work=90.502992 tx=397738854 date='2019-04-02T10:30:13Z' progress=1.000000 cache=19.1MiB(168908txo) warning='43 of last 100 blocks have unexpected version'
2019-04-02T10:32:28Z UpdateTip: new best=000000000000000000230bbf6cdf5920eebf0a607aa06216b95d1a7f8b70f1e8 height=569867 version=0x20000000 log2_work=90.503014 tx=397741768 date='2019-04-02T10:31:58Z' progress=1.000000 cache=19.6MiB(173867txo) warning='43 of last 100 blocks have unexpected version'
2019-04-02T10:32:30Z Pre-allocating up to position 0x300000 in rev01582.dat
2019-04-02T10:32:31Z UpdateTip: new best=00000000000000000018eda4c4f51c471b3232fd842d60106897a59829ce5c48 height=569868 version=0x20c00000 log2_work=90.503037 tx=397744157 date='2019-04-02T10:32:26Z' progress=1.000000 cache=19.9MiB(176585txo) warning='43 of last 100 blocks have unexpected version'
2019-04-02T10:35:33Z Pre-allocating up to position 0x2000000 in blk01582.dat
2019-04-02T10:35:34Z UpdateTip: new best=0000000000000000002a9eaa22dbfd5a8bbe26ebf324ee93e2823537bf5f56c4 height=569869 version=0x20c00000 log2_work=90.503059 tx=397746876 date='2019-04-02T10:35:01Z' progress=1.000000 cache=20.3MiB(180054txo) warning='44 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
[... cut ...]
2019-04-02 11:50:10.008 [DBG] PEER: Received ChannelAnnouncement(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605545034027040768) from 104.248.84.249:9735
2019-04-02 11:50:10.009 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=622736997815549952, mflags=00000000, cflags=00000010, update_time=2019-04-02 11:12:08 +0100 BST) from 54.37.22.216:9735

*** LAST 20 RTL LOGS ***
-- No entries --

rootzoll, 2019-04-02:
Your LND nodes seems to crash on this logs:

```
Apr 02 11:50:11 FreeTradeBulgaria lnd[12459]: /Users/roasbeef/gocode/pkg/mod/github.com/btcsuite/btcwallet@v0.0.0-20190123033236-ba03278a64bc/wallet/wallet.go:186 +0x124
Apr 02 11:50:11 FreeTradeBulgaria lnd[12459]: goroutine 115 [IO wait]:
Apr 02 11:50:11 FreeTradeBulgaria systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
```

From v1.0 to v1.1 LND and bitcoid version stayed the same ... so I dont think its related to the update in particular. Maybe just to update put the LND into some situation (booting, catchingup, etc) that triggered this case.

Because this is an LND error that I did not seen before, I would recommend you to go over to the LND dev Slack or the LND issues and see if you there can get some feedback on: https://dev.lightning.community

vamp111, 2019-04-02:
Thx Chris, will forward these to LND slack

closed 2019-04-02 by rootzoll