raspiblitz/raspiblitz #484: Mobile wallet connection Impriovements

rootzoll, 2019-04-02:
On next release test again the mobile wallet connection processes. Had some issue with latest shango download in local network and also recheck issues like tose ones:
https://github.com/rootzoll/raspiblitz/issues/476

Also if possible copy over the LCD display QR code ... the QR code zoom in/out is really a pain.

rootzoll, 2019-04-03:
Check and replace forwarded ports and domains for QRcode info.

openoms, 2019-04-03:
Noting that Shango was last updated in 2018 Dec 8 and still early access release.

rootzoll, 2019-04-06:
I will keep Shango as a option for now, but Zeus ist now also part of the list.

rootzoll, 2019-04-27:
OK Wallet connection looks good so far for v1.2 - closing issue.

closed 2019-04-27 by rootzoll