raspiblitz/raspiblitz #485: forwarding ports by autossh tunnel

rootzoll, 2019-04-02:
A method for people that rae behind a NAT and cannot change that to make ports public by a ssh reverse tunnel using a public sshd server.

cryptomulde, 2019-04-03:
noice!

rootzoll, 2019-04-09:
- test port forwarding of LND 9735
- test with more then 3 forwardings

rootzoll, 2019-04-27:
See FAQ: How to setup port-forwarding with a SSH tunnel?

rootzoll, 2019-04-28:
OK. Its working for v1.2. Closing issue.

frennkie, 2020-07-23:
I'm having issues with this. See #1194 

This does not survive a DSL reconnect with a changed IP.

```
ExecStart=/usr/bin/autossh -M 0 -N -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no -o ServerAliveInterval=60 -o ServerAliveCountMax=2 [PLACEHOLDER]
```

This does:

```
ExecStart=/usr/bin/autossh -M 20000 -N -o UserKnownHostsFile=/dev/null -o StrictHostKeyChecking=no -o ServerAliveInterval=60 -o ServerAliveCountMax=2 [PLACEHOLDER]
```

rootzoll, 2020-08-08:
- [ ] before ports add an optional parameter for m ... for example "--m:20000"

rootzoll, 2020-09-10:
tested: on recovery the `sudo cat /etc/systemd/system/autossh-tunnel.service` shows that it regenerates the optional monitoring parameter in the service file

rootzoll, 2020-10-08:
OK. Closing this for the v1.6.1 release.

closed 2020-10-08 by rootzoll