raspiblitz/raspiblitz #489: Commit / Commitment Fees?

funk256, 2019-04-04:
Not sure this would really be an "issue" so much as an open question about the lightning network itself.  I tend to have a hard time finding straight answers on Google.  Can't tell if that's because the lightning network itself is still a pretty new thing or my Google skills aren't as good as I thought.

In any case, I have been checking my raspiblitz occasionally to make sure it's still online and everything is going smoothly.  I tend to use the status screen or go into the detailed wallet balances option to see the total number of Satoshis and what not.  I had noticed my total number was going down very slightly so I looked into it a bit deeper.

On the wallet screen (I think the open channels screen too) it shows my total and the amount I have sent to apps on the lightning network itself and that part checks out.  There is another option I'm unfamiliar with, though.  The section for commitment fees.  Can anyone explain how that works?  I wish I had tracked it a lot closer now but seems like it kept going up to a higher number over days even though I wasn't making any transactions or doing anything on my node?

Maybe that's not quite right.  I'll keep looking into it but just wasn't expecting anything like that one way or the other.  Can I expect that to keep going up, does it go off of how many Satoshis you have locked up in a channel or what exactly?  I had assumed once you had funds locked up in a channel the only thing that would change your final amount would be if you went in and made transactions but that's just an assumption.  Definitely open minded to any new info someone can present to me.

If it might help this is the only node I have an open channel with:  https://1ml.com/node/02529db69fd2ebd3126fb66fafa234fc3544477a23d509fe93ed229bb0e92e4fb8  Thanks again.

CommanderPoe, 2019-04-04:
Hey mate, as you said this might not be the right place to ask these kind of questions but as far as im concerned those fees are reserved in case your partner is uncoperative and you want to force-close the channel, so they are likely to vary up/down according to the mempool but that is not the only metric. Closing channels uncooperatively might be more expensive than cooperatively. 

I recommend you to join the lightning network channel on telegram, there you can ask all sort of question you like, there are very helpful people around, tho dont get too scared since lately they are allowing ton of bcashers to spam the channel... such a pity.

funk256, 2019-04-04:
First of all, thanks for your reply.

That's a really good idea, actually. Was struggling with places to reach out to outside of this. I'll probably give that a try. Your explanation on the commitment fees make sense and that does help. Just goes to show there's still a lot more I can learn as far as all this goes.

CommanderPoe, 2019-04-05:
Exactly we are still very early and thers's still lot to leant and understand... bitcoin will never be finished is a forever entrepeunorial form of life.

Now please close the issue, im sure rootzoll wanna keep this as clean as possible

closed 2019-04-06 by rootzoll