raspiblitz/raspiblitz #491: problem: lcd_rotate=2 (rotate my Raspiblitz LCD 180°)

Dimpelmoser, 2019-04-04:
Hello and sorry, I am a beginner as well in raspiblitz as in Linux as on Github.

I succcesfully installed a Raspiblitz (first v.1.0, then v1.1) and I love it.

Now I wanted to rotate my raspiblitz-3,5'LCD.

In the www I found: https://www.rasppishop.de/Das-Raspberry-Pi-7-Display-180-Grad-drehen-So-gehts

Then I did via PUTTY on my raspiblitz-main-menue
X for the console/termonal

There the command:
sudo nano /boot/config.txt

At the end of the displayed config.txt I added:
lcd_rotate=2
btw
lcd_rotate = 2
btw
display_rotate=2
btw
display_rotate = 2

Every single attempt I saved the result and reboot my raspiblitz ... but without the desired result.

Sorry please for the probably stupid question, but is there any help for me - Thanks.

openoms, 2019-04-05:
Try `display_rotate=1` or `display_rotate=3`
or see this: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#can-i-flip-the-screen

Dimpelmoser, 2019-04-05:
@openoms: in your posted link https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#can-i-flip-the-screen i found my desired solution:
In /root/config.txt I changed "dtoverlay=tft35a:rotate=" from 270 in 90, and after a reboot the LCD-screen of my raspiblitz had turned 180 degrees.
Thank you very, very much for your help!

closed 2019-04-05 by Dimpelmoser