raspiblitz/raspiblitz #492: Progress Freezes at "Lightning Filtering Blockchain"

felipeclopes, 2019-04-04:
I've seem many other issues about the same behavior, but none of them seemed to help me get this fixed. I got this in the `v1.0` and then upgraded my SSD card to the `v1.1` and it seems to present the same behavior.

``` shell
/home/admin/XXdebugLogs.sh

*** RASPIBLITZ LOGS ***
blitzversion: 1.1
chainnetwork: bitcoin / main
 22:38:08 up 3 min,  2 users,  load average: 2.60, 1.11, 0.43

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-04-04 22:35:58 BST; 2min 9s ago
  Process: 1004 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1022 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─1022 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Apr 04 22:35:58 faisca systemd[1]: Starting Bitcoin daemon...
Apr 04 22:35:58 faisca systemd[1]: Started Bitcoin daemon.

*** LAST 5 ERROR LOGS ***
-- Logs begin at Thu 2019-04-04 22:35:00 BST, end at Thu 2019-04-04 22:38:08 BST. --
Apr 04 22:35:58 faisca systemd[1]: Starting Bitcoin daemon...
Apr 04 22:35:58 faisca systemd[1]: Started Bitcoin daemon.

*** LAST 20 INFO LOGS ***
2019-04-04T21:37:43Z Bound to 0.0.0.0:8333
2019-04-04T21:37:43Z init message: Loading P2P addresses...
2019-04-04T21:37:43Z Leaving InitialBlockDownload (latching to false)
2019-04-04T21:37:44Z Loaded 51073 addresses from peers.dat  1309ms
2019-04-04T21:37:44Z init message: Loading banlist...
2019-04-04T21:37:44Z init message: Starting network threads...
2019-04-04T21:37:44Z net thread start
2019-04-04T21:37:44Z dnsseed thread start
2019-04-04T21:37:44Z addcon thread start
2019-04-04T21:37:44Z init message: Done loading
2019-04-04T21:37:44Z msghand thread start
2019-04-04T21:37:44Z opencon thread start
2019-04-04T21:37:53Z txindex is enabled at height 570217
2019-04-04T21:37:53Z txindex thread exit
2019-04-04T21:37:54Z New outbound peer connected: version: 70015, blocks=570217, peer=0
2019-04-04T21:37:55Z Loading addresses from DNS seeds (could take a while)
2019-04-04T21:38:01Z New outbound peer connected: version: 70015, blocks=570217, peer=1
2019-04-04T21:38:03Z 239 addresses found from DNS seeds
2019-04-04T21:38:03Z dnsseed thread exit
2019-04-04T21:38:08Z New outbound peer connected: version: 70015, blocks=570218, peer=2

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-04-04 22:35:58 BST; 2min 9s ago
 Main PID: 1023 (lnd)
   CGroup: /system.slice/lnd.service
           └─1023 /usr/local/bin/lnd --externalip=73.189.155.176

Apr 04 22:35:58 <user> systemd[1]: Started LND Lightning Daemon.

*** LAST 5 LND ERROR LOGS ***
-- Logs begin at Thu 2019-04-04 22:35:00 BST, end at Thu 2019-04-04 22:38:08 BST. --
Apr 04 22:35:58 faisca systemd[1]: Started LND Lightning Daemon.

*** LAST 20 LND INFO LOGS ***
2019-03-31 18:50:21.053 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-03-31 18:50:21.053 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-03-31 18:50:21.385 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-03-31 18:50:21.385 [INF] RPCS: Generating TLS certificates...
2019-03-31 18:50:21.395 [INF] RPCS: Done generating TLS certificates
2019-03-31 18:50:21.402 [INF] RPCS: password RPC server listening on [::]:10009
2019-03-31 18:50:21.402 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-03-31 18:50:21.402 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-04-04 15:27:19.422 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-04-04 15:27:19.422 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-04-04 15:27:19.517 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-04 15:27:19.548 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-04 15:27:19.548 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-04 15:27:19.548 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2019-04-04 22:35:59.817 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default
2019-04-04 22:35:59.817 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-04-04 22:35:59.913 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-04 22:35:59.945 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-04 22:35:59.952 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-04 22:35:59.952 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
```

After trying this I rebooted the raspiblitz with the command: `sudo shutdown -r now`

It got stuck in the same stage, so I went to check the config files and the one thing that I noticed is that they have still the `raspiBlitzVersion='1.0'` which seems odd, since I upgraded to `1.1`.

``` shell
$ sudo cat /home/admin/raspiblitz.info
state=ready
message='waiting login'
network=bitcoin
chain=main
setupStep=100
undervoltageReports=0
```
``` shell
$ sudo cat /mnt/hdd/raspiblitz.conf
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.0'
network=bitcoin
chain=main
hostname=faisca
publicIP=73.189.155.176
autoPilot=off
autoNatDiscovery=off
runBehindTor=off
rtlWebinterface=off
```
``` shell
$ cat _version.info 
# RaspiBlitz Version - always [main].[sub]
```
``` shell
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  2.8G   11G  21% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M  6.3M  458M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   44M   22M   22M  51% /boot
/dev/sda1       916G  274G  596G  32% /mnt/hdd
tmpfs            93M     0   93M   0% /run/user/1000
tmpfs            93M     0   93M   0% /run/user/1001
```

The following troubleshoot step that I tried was to create the wallet myself, which seemed to be successfull untill I could unlock the wallet that I just created.

``` shell
$ lncli create
Input wallet password: 
Confirm wallet password: 

Do you have an existing cipher seed mnemonic you want to use? (Enter y/n): n

Your cipher seed can optionally be encrypted.
Input your passphrase if you wish to encrypt it (or press enter to proceed without a cipher seed passphrase): 
Confirm cipher seed passphrase: 

Generating fresh cipher seed...

!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!

---------------BEGIN LND CIPHER SEED---------------
 1. ... etc
---------------END LND CIPHER SEED-----------------

!!!YOU MUST WRITE DOWN THIS SEED TO BE ABLE TO RESTORE THE WALLET!!!
[lncli] rpc error: code = Unknown desc = password must have at least 8 characters
```

Unfortunately:

``` shell
lncli getinfo
[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

$ lncli unlock
Input wallet password: 
[lncli] rpc error: code = Unknown desc = wallet not found
```

I restarted again with no luck. 

Any ideas on how can I troubleshoot further or what should be happening with the node?

Luigybruno, 2019-04-06:
I had the same problem starting from scratch with 1.1
Then I started all over again with 1.0 and it seems to be fine.

Luigybruno, 2019-04-06:
try ./70initLND.sh

felipeclopes, 2019-04-06:
I tried already, no luck. I'll do a fresh install and clean up my HDD, maybe it is a corrupted blockchain or something of the sorts. I'll report progress.

zuardi, 2019-04-06:
Guys, I'm +1 with this problem.
My old RaspiBlitz was faulty. I back the LND files up and did a fresh install. It was working properly.
Then I restored the LND files through `/home/admin/config.scripts/lnd.rescue.sh`
Now I am stuck in a start/close LND loop.
https://termbin.com/g5xk
https://termbin.com/u4do

Maybe has something to do with the restored files...

fluidvoice, 2019-04-09:
It might need to open an issue with LND. I don't know what this means in the log:
[ERR] DISC: Unable to rebroadcast stale channels: unable to retrieve outgoing channels: edge not found

felipeclopes, 2019-04-09:
@fluidvoice, I'm starting with a fresh start, since I didn't have a wallet setup yet. 

If it happens again, I'll reach out to LND team and update the Issue accordingly.

openoms, 2019-04-10:
Anyone stuck on this error can you please try:
`sudo -u bitcoin /usr/local/bin/lncli getinfo`

if this shows a permission error, you have the same bug as me. I managed to solve it by deleting `-u bitcoin` from the 00mainMenu.sh lines 248 and 253. I make you a helper script or more detailed instructions if it is the same problem (I am building on OdroidXU4 with a minimal Ubuntu image, so things might be different)

fluidvoice, 2019-04-10:
> It might need to open an issue with LND. I don't know what this means in the log:
> [ERR] DISC: Unable to rebroadcast stale channels: unable to retrieve outgoing channels: edge not found

I didn't notice in the log before but LND is restarting (lots of times) after this error.  I might ping the LND dev's to ask about this. What/why this is happening.

> 2019-04-06 22:46:25.829 [ERR] DISC: Unable to rebroadcast stale channels: unable to retrieve outgoing channels: edge not found
> 2019-04-06 22:46:25.870 [INF] LTND: Shutdown complete
2019-04-06 22:47:26.423 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default

zuardi, 2019-04-10:
Thanks, will open a issue there!

> > It might need to open an issue with LND. I don't know what this means in the log:
> > [ERR] DISC: Unable to rebroadcast stale channels: unable to retrieve outgoing channels: edge not found
> 
> I didn't notice in the log before but LND is restarting repeatedly after this error. I might ping the LND dev's to ask about this. What/why this is happening. LND is in a bad state for some reason?
> 
> > 2019-04-06 22:46:25.829 [ERR] DISC: Unable to rebroadcast stale channels: unable to retrieve outgoing channels: edge not found
> > 2019-04-06 22:46:25.870 [INF] LTND: Shutdown complete
> > 2019-04-06 22:47:26.423 [INF] LTND: Version: 0.5.2-beta commit=v0.5.2-beta, build=production, logging=default

fluidvoice, 2019-04-10:
@zuardi copy the link (https://github.com/rootzoll/raspiblitz/issues/492#issuecomment-481807023) of your above issue and paste it into the LND issue so we can see/refer to it.

nyxnor, 2021-05-17:
Please close this issue.

closed 2021-05-17 by felipeclopes