raspiblitz/raspiblitz #493: LCD has inverted color (incorrect driver Waveshare!)

intorid, 2019-04-04:
Hi,
I installed my Raspberry pi 3 B+ [LCD 3.5inch RPi LCD (B) from WaveShare SpotPear revision OLD](https://imgur.com/0EYhFNy) 
the [WaveShare reports released](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(B)) from the production stop, a new version of screens has been.

**Additionally, I note that I use additional power (5.1V 4A for SSD and 5V 3,5A for the RPi + LCD) low voltage log problems caused by a startup experiment from a single power supply**

> Version Update
> Because of stop production of old LCD, we update 3.5inch RPi LCD (B) to Rev2.0
> The dimension of LCD is different, and driver is updated as well.(Please download the new driver if you have the new version)
> You can recognize new and old version by the silk screen printing on the backside. The new version has screen printing Rev2.0

After deploying RaspiBlitz version 1.1, I get a completely [inverted screen color](https://imgur.com/0Ek9f2e)

Next step reinstalling the driver for the OLD version according to the [instructions WaveShare](http://www.waveshare.net/wiki/3.5inch_RPi_LCD_(B)). 

> git clone https://github.com/waveshare/LCD-show.git
> .cd LCD-show/
> If it is an OLD revision driver, execute the following code:
> sudo ./LCD35B-show
> If it is a NEW revision driver, execute the following code：
> sudo ./LCD35B-show-V2

I got normal colors, but the start [screen is a double problem](https://imgur.com/u4BnZ0q)
**What else do I need to do to return the normal display of the start screen?**

intorid, 2019-04-05:
Purchase shop [Gearbest](https://www.gearbest.com/diy-parts-components/pp_663526.html)

openoms, 2019-04-05:
To sort out the double status screen problem you need to increase the fontsize.
Here is how to do it: https://github.com/rootzoll/raspiblitz/issues/244#issuecomment-476713706

intorid, 2019-04-05:
> To sort out the double status screen problem you need to increase the fontsize.
> Here is how to do it: [#244 (comment)](https://github.com/rootzoll/raspiblitz/issues/244#issuecomment-476713706)

Thanks, openoms! I set it Fixed and I managed to fix the double screen problem. Charm works

closed 2019-04-05 by intorid