raspiblitz/raspiblitz #494: Loop of LND Unlock, Filtering Blockchain

wuestenigel, 2019-04-05:
Hey guys,

I'm struggling for some months about one problem:

Whenever I want to do some stuff in the main menu (like transferring my onchain funds), I'm prompeted to insert my LND password. After that, some strange block filtering starts with 570323/570323 and wait and inserting LND password again and again and again.
*SOMETIMES* after  this, I may get back to the main menu and I can select an option. But then everything starts again.

What I already did:
- upgraded from 0.9 to 1.0 and waited about 5 weeks to verify
- upgraded from 1.0 to 1.0
- replaced power supply with the recommended one from the main post about raspiblitz
- tried some commands from #292 

I'm totally lost. I only want to get my funds and then maybe start with a smaller amount of BTC and get the device running.

Thank you so much guys for any kind of help

[image]
[image]
[image]
[image]

rootzoll, 2019-04-06:
A Unlock-Loop is happening, when LND is crashing and the RaspiBlitz tries to start it up again. On every re-startup after a crash, you are asked to unlock your wallet again.

The question is why LND is crashing. If we look at the logs you posted (under LND Error Logs), there is "LND panic invalid page type 13434: 10" and last normal log around that time was "syncing channel graph" that seems to be the reason. But I have no idea what that is. That seems to be a LND problem. So it would be best to ask at the LND Dev Slack or GitHub for more details: https://dev.lightning.community

Possibly related LND GitHub issues could be those: https://github.com/LightningNetwork/lnd/issues?utf8=%E2%9C%93&q=is%3Aissue+panic+sync+channel+graph

But we are preparing the new RaspiBlitz v1.2 release which will have the new LND 0.60 on board (once the final release is out). There is a change that this is fixed than, but thats not 100%.

wuestenigel, 2019-04-15:
Thanks, I'm patient to the next cool versions of raspiblitz.

 Is there at least a way to load my funds from the Wallet? (they are onchain, not LND)

fluidvoice, 2019-04-15:
> A Unlock-Loop is happening, when LND is crashing and the RaspiBlitz tries to start it up again. On every re-startup after a crash, you are asked to unlock your wallet again.
>
> The question is why LND [...]

A similar LND panic was reported here: https://github.com/lightningnetwork/lnd/issues/2626
And @Roasbeef said 

> "Looks like rasp pi disk corruption."

So you might want to try testing your power supply to make sure it's adequate, and perhaps buy a better new SD card and burn it and start anew.

wuestenigel, 2019-05-04:
Hey,
I'm in the loop again after upgrading to v1.2

[image]
[image]
[image]

Any ideas how I can fix this? Is there any way to release my 0.05 BTC (onchain)?

fluidvoice, 2019-05-04:
Please provide links to these full logs.
syslog: `sudo cat /var/log/syslog | nc termbin 9999`
bitcoind: `sudo cat /mnt/hdd/bitcoin/debug.log | nc termbin.com 9999`
lnd: `sudo cat /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log | nc termbin.com 9999`

fluidvoice, 2019-05-04:
In the meantime there is a recent LND release in the past 24 hours that has some bug fixes. So perhaps we can apply it and see if it fixes things for you?
https://github.com/lightningnetwork/lnd/releases/tag/v0.6.1-beta-rc1

wuestenigel, 2019-05-04:
@fluidvoice thanks!

raspiblitz: https://termbin.com/l68f
syslog: https://termbin.com/ec8e
bitcoind: https://termbin.com/6xt4
lnd: https://termbin.com/pwwu

fluidvoice, 2019-05-04:
@wuestenigel I would try burning the SD card and trying again to see if the problem is reproducible or perhaps something bad on disk.

wuestenigel, 2019-05-04:
@fluidvoice Thanks for the idea. I'm struggling with this for months since v0.9.

Completely changed my power supply and SD Card and flashed it like 10x times.

I just want to release my funds so I'm safe and may start from scratch again. :/

fluidvoice, 2019-05-04:
> @fluidvoice Thanks for the idea. I'm struggling with this for months since v0.9.
> 
> Completely changed my power supply and SD Card and flashed it like 10x times.
> 
> I just want to release my funds so I'm safe and may start from scratch again. :/

Oh, I didn't realize this was the same problem (lnd panic) as before.  Very strange as nobody else is seeing this.

fluidvoice, 2019-05-07:
> @fluidvoice Thanks for the idea. I'm struggling with this for months since v0.9.
> 
> Completely changed my power supply and SD Card and flashed it like 10x times.
> 
> I just want to release my funds so I'm safe and may start from scratch again. :/

@wuestenigel A LND dev said about your issue **"it seems like you've encountered database corruption."**

wuestenigel, 2019-05-07:
@fluidvoice Thanks for your help. Problem is this may take another weeks or months(!!!) to download the data again. I had this experience already.

Do you know if it may be ok to delete everything without the blockchain data?

fluidvoice, 2019-05-07:
> @fluidvoice Thanks for your help. Problem is this may take another weeks or months(!!!) to download the data again. I had this experience already.
> 
> Do you know if it may be ok to delete everything without the blockchain data?

You could try it.  Delete only the LND data (**/mnt/hdd/lnd**) and burn a new SD card, while keeping the blockchain data (**/mnt/hdd/bitcoin**).  It might work but might not. I'd prob try it...  

`sudo systemctl stop lnd.service`
`sudo systemctl stop bitcoind.service`
`sudo rm -rf /mnt/hdd/lnd`
`sudo shutdown now`
now burn new SD card and reinstall.

btw, I too got frustrated with delays in downloading/syncing the blockchain so now I made a 2nd copy as a backup on an old spare 3.5" HDD.  No more weeks of downloading for me!

fluidvoice, 2019-05-07:
wait...

fluidvoice, 2019-05-07:
it doesn't work by default that way. I tried it.  You cannot remove the entire directory.  Let me see how to remove only the data portion of that directory.

wuestenigel, 2019-05-07:
Hey @fluidvoice,

great. Should I rename now or just wait until you experimented a little bit?

Very much appreciated!

Marco

fluidvoice, 2019-05-07:
> Hey @fluidvoice,
> 
> great. Should I rename now or just wait until you experimented a little bit?
> 
> Very much appreciated!
> 
> Marco

you can restore via the seed phrase when you first created your LND wallet. For me no. I cannot find my seed phrase but no big deal as only a tiny amount of $ was in it.

wuestenigel, 2019-05-08:
@fluidvoice 

I tried to rename the folder, but after creating a new SD card it says:

./00raspiblitz.sh: line 336: [: -eq: unary operator expected
WARN: LND needs re-setup
*** Check Basic Config ***
*** Check bitcoin Running ***
*** Check bitcoin Responding *** (can take a while)
OK - chainnetwork is working

*** Check LND Config ***
cp: cannot create regular file '/mnt/hdd/lnd/lnd.conf': No such file or directory
sed: can't read /mnt/hdd/lnd/lnd.conf: No such file or directory
chown: cannot access '/mnt/hdd/lnd/lnd.conf': No such file or directory
FAIL - Was not able to setup LND
admin@wuestenigelRaspi:~ $

wuestenigel, 2019-05-13:
[image]

Working again!!! There is nothing as good as a clean install. Thanks!!

closed 2019-05-13 by wuestenigel