raspiblitz/raspiblitz #495: Unable to receive payment even with direct channel and remote balance V1 onwards

walletofsatoshi, 2019-04-06:
I was able to receive payment in version 0.9 raspiblitz  but after v1 unable to receive payment even with direct channel and remote balance on direct channel. Any fixes?

rootzoll, 2019-04-06:
And this should still be possible. What is the error message when you try to send it?

walletofsatoshi, 2019-04-06:
There was problem processing this payment
Please check address and try again.

Tried generating invoice both via ssh and zap wallet. Able to send but not received. It worked fine v0.9

rootzoll, 2019-04-06:
```
There was problem processing this payment
Please check address and try again.
```
Which software is giving you that error message? Is there a way to get a more technical error message ... like when sending from lncli?

walletofsatoshi, 2019-04-06:
It was from wallet of Satoshi also tried BlueWallet. Both failed and I could not receive to my raspiblitz. I don’t know a way to get incli from them.

rootzoll, 2019-04-06:
Without a proper error message its hard to tell whats the problem in your case. But I can say that I just tested receiving payments with the RaspiBlitz just recently and in general its working.

walletofsatoshi, 2019-04-06:
Can you try wallet of Satoshi or BlueWallet and confirm it’s working ? Do you recommend any nodes that I can connect which can help with this?

walletofsatoshi, 2019-04-09:
Solution was spending over 1% of your channel capacity. So if you spend 3% you can receive upto 2%

closed 2019-04-09 by walletofsatoshi