raspiblitz/raspiblitz #496: RFC: research proper power on/off methods

fluidvoice, 2019-04-06:
Opening an issue to discuss and keep track of possible on/off power switches and methods since powering down or yanking the micro-usb power cable without first doing a `sudo shutdown now` can cause file corruption.
I found cheap product called a GPIO "shim" that still allows connecting the LCD here:
P  (Pinout diagram here)

Won't work with large heatsinks like geekworm passive heatsink here:
_Originally posted by @rootzoll in https://github.com/rootzoll/raspiblitz/issues/474#issuecomment-480514649_

fluidvoice, 2019-04-06:
Pic shows it soldered to a riser, but it could be soldered to the GPIO pins without the riser allowing the LCD to still be plugged on top.  

[image]

rootzoll, 2019-04-06:
We have a touch LCD already (its just inactive and not used at the moment)- if someone can get it to work we would not need extra hardware for interaction like power-off.

fluidvoice, 2019-04-06:
> We have a touch LCD already (its just inactive and not used at the moment)- if someone can get it to work we would not need extra hardware for interaction like power-off.

Yeah, I was thinking it should be possible to do this in software. It's why I titled this "methods" ;-)
Needs more research how to do it.

WetPuppyKisses, 2019-04-08:
The ideal would be a mini DC UPS with an auxiliary dry contact for low battery. When the battery is low, it send a signal to the raspberry and this signal is interpreted as a shutdown command

frennkie, 2019-05-05:
> We have a touch LCD already (its just inactive and not used at the moment)- if someone can get it to work we would not need extra hardware for interaction like power-off.

@rootzoll Getting the touch LCD to work is really easy on the graphically Raspbian desktop. I have unsuccessfully tried to find a way to make use of touch on the non-X CLI in the past. So today I (again) looked into taking the "GUI" path.

I think there are two different ways to approach this:
* 1) Doing a full rewrite of the current (dialog/whiptail-based) menus and interfaces in a GUI framework (GTK, QT, ...)
* 2) Using a small GUI wrapper that has limited touch space and re-uses the existing menus and interfaces. 

Option i) needs experience and - I assume - will be a lot of work. Option ii) has the benefit that the known user experience is mostly preserved. 

I put together a really quick&dirty PoC that has four touch buttons (PEN or finger) on the left and the rest of the screen shows a "XTerm" window (running `00infoLCD.sh`) [GIF](https://media.giphy.com/media/RlOAgL1ttgRYbsdqyq/giphy.gif):

[image]

Here is a short write-up, let me know what you think:
https://gist.github.com/frennkie/4d99cb35a3c62033a535564220c11150

fluidvoice, 2019-05-05:
> quick&dirty PoC that has four touch buttons (PEN or finger) on the left and the rest of the screen shows a "XTerm" window

will these buttons be seen and selectible via mouse and/or keyboard from an SSH session?

frennkie, 2019-05-05:
In short: No.

The idea here is to make use of the existing touch screen of the Raspiblitz. Primarily to do a few (1-3) of the main things that people might want to do without having to use another computer -  e. g. a clean shutdown.
As this uses the Xserver/X.org you could actually start the GUI wrapper remotely (as long as you have a local X server running on the second computer). But this is not the intended use case.

rootzoll, 2019-05-13:
@frennkie much thanks for going deeper into this :) I marked it to be part of a future release - I think after v1.3, because first I have still some basic things to fix first (data protection, stability, version updates). But then activating touch on the LCD to give point on my wish list.

nyxnor, 2021-04-18:
please close this. Openoms solved with in the _command `restart`.

closed 2021-04-19 by rootzoll