raspiblitz/raspiblitz #497: Unable to cashout

StifflersMoM69, 2019-04-06:
Hi!
I get this error, altough there are enough sats.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
FINAL FAIL --> Was not able to send transaction with max 20000 sat
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Press ENTER to return to main menu.

rootzoll, 2019-04-08:
Have you tried the RTL web UI (can be activated under services) to make that transaction? Its way better then the shell script on making transactions.

StifflersMoM69, 2019-04-08:
There i can only do Lightning Payments or am i able to send to any BTC address?

rootzoll, 2019-04-08:
With RTL you can manage your on-chain and off-chain funds. See under "LND Wallet" there is a send option for BTC.

rootzoll, 2019-04-08:
TODO: RTL has a sweep all option when sending BTC funds. Check how they do it and copy that for the shell script.

StifflersMoM69, 2019-04-08:
Sending Fund Failed
Code
:
500
Message
:
error:insufficient funds available to construct transaction
Code
:
2

rootzoll, 2019-04-08:
It may have something todo with fees .. maybe start with sending half the funds amount out and then check how much fees you need to set to make it work (raise setup by step)

rootzoll, 2019-04-08:
Also check in the RTL UI if all your channels are closed and setteled. If you have a force closing still running some funds may not be spendale already.

StifflersMoM69, 2019-04-08:
sending half of funds worked. The fees are much higher then expected

CommanderPoe, 2019-04-09:
> sending half of funds worked. The fees are much higher then expected

Mempool is been steadily clearing out, but fees are indeed higher than a month ago.

rootzoll, 2019-04-12:
TODO: Improve the CashOut Script - python rewrite.

rootzoll, 2019-04-15:
TODO: Test the new --sweepall feature of LND 0.6 with RaspiBlitz v1.2

rootzoll, 2019-04-15:
The new CASHOUT in v1.2 is much simpler now to the LND 0.6 sendcoins --sweepall feature. No rewrite in python needed. 

TODO: Final testing

rootzoll, 2019-04-15:
OK. The two tests worked great. I think this improves the CASHOUT experience by a lot. Closing Issue.

closed 2019-04-15 by rootzoll