raspiblitz/raspiblitz #500: Cipher seed bip39 not compatible with other wallets-critical issue

walletofsatoshi, 2019-04-09:
I tried recovering my on chain wallet with 24 raspiblitz word seed on electrum . It failed to recover my on chain funds. I think this is a critical issue. The checksum failed. I do not believe the seed is generated properly. Until this is resolved users are at significant risk of loosing funds.

CommanderPoe, 2019-04-09:
Lightning per se is very experimental and users have always been at significant risk of losing their funds, nothing new to see here, however ( i have not tried yet ) but as far as im concerned you need the wallet db directory of your lnd in order to recover on chain funds properly i read somewhere?

 Maybe someone can correct me on this one but last time i tried to recover my on chain funds after i migrated from Raspibolt to Raspiblitz the only way to recover them was bringing on that old wallet db.  By simply using the cipher seed you wont recover your funds, i tried several times with different heights but it never worked out. only when i brought that file, funds started to appearing steadily.

walletofsatoshi, 2019-04-09:
Thanks. I understand the experimental nature of lightning at the moment. But till the seed is recoverable, it should be clearly stated in the read me document that this won’t work on any other wallets and if you believe wallet.db is key to funds recovery it should be mentioned clearly in the user guide. But work needs to be done as priority to make this seed recoverable in any bip39 compatible wallets if we really want to make this project a huge success.

rootzoll, 2019-04-09:
@walletofsatoshi RaspiBlitz is a open source project. Feel free to make a pull request containing the information added you think that is still missing during RaspiBlitz Setup. 

Providing a stable and compatible seed recovery needs to be part of LND - please report over at the LND Dev Slack or GitHub what features you are still missing and are important to you: https://dev.lightning.community

rootzoll, 2019-04-09:
@walletofsatoshi of course if you are not a programmer its always great if you outline a content change (text change) more concrete - like: please exchange this text here with this text suggestion at that exact piont in the process.

walletofsatoshi, 2019-04-18:
Checked on slacks with Devs they recommend seed recovery by running lnd recovery by hand not by raspiblitz script guide by roasbeef

rootzoll, 2019-04-18:
At the moment the RaspiBlitz is using/calling the lncli wallet create and
is letting the user answere the LND questions to revover from seed. Did
devs on slack mentioned what the RaspiBlitz is doing different compared
doing it by hand?

walletofsatoshi, 2019-04-18:
Yes that’s what they implied. Also wanted to let you know that the seed is not bip39 hence will not work on other bip39 wallets

fluidvoice, 2019-04-18:
> the seed is not bip39

Please explain in detail

> Checked on slacks with Devs they recommend seed recovery by running lnd recovery by hand 

Please explain in detail what you do "by hand"?

fluidvoice, 2019-04-18:
> At the moment the RaspiBlitz is using/calling the lncli wallet create and is letting the user answere the LND questions to revover from seed. Did devs on slack mentioned what the RaspiBlitz is doing [...]

@rootzoll this LND commit might help:
https://github.com/lightningnetwork/lnd/commit/dd48a36226521f6e6e2dddc456c4c0cf3c5d48ea

walletofsatoshi, 2019-04-18:
Yes they asked to follow this

the recovery seed while creating lnd wallet is not bip39 will not restore on chain funds in any bio 39 compatible wallets like electrum.  You have to follow this roasbeef or similar lnd recovery guide without raspiblitz which would be manual

fluidvoice, 2019-04-18:
According to roasbeef's comments in the commit doc, BIP39 is inferior.  But also looks like the cipher seed is optional...

> Users can also _optionally_ enter a second passphrase which we call the _cipher
> seed passphrase_:
> ```
> Your cipher seed can optionally be encrypted.
> Input your passphrase if you wish to encrypt it (or press enter to proceed without a cipher seed passphrase):
> ```
> 
> If specified, then this will be used to encrypt the cipher seed itself. The
> cipher seed format is unique in that the 24-word phrase is actually a
> _ciphertext_. As a result, there's no standard word list as any arbitrary
> encoding can be used. If a passphrase is specified, then the cipher seed you
> write down is actually an _encryption_ of the entropy used to generate the BIP
> 32 root key for the wallet. **Unlike a BIP 39 24-word phrase, the cipher seed is**
> **able to _detect_ incorrect passphrase. BIP 39 on the other hand, will instead**
> **silently decrypt to a new (likely empty) wallet.**
> 
>

closed 2019-04-24 by walletofsatoshi

rootzoll, 2019-04-28:
Just some follow up info for archive readers: The Lightning LND Desktop App was released: https://github.com/lightninglabs/lightning-app/releases and I heard good feedback on using it to recover on-chain funds from the LND wallet seed. So check it out if you are on a quest to recover your funds from seed.