raspiblitz/raspiblitz #502: stop using SCP, use rsync instead

fluidvoice, 2019-04-09:
Is SCP unreliable? It's possible (see below).  I read that 
> **scp does not guarantee file integrity since it does not use file checksums like `rsync` does.**

so potentially Raspiblitz should stop using SCP.  Needs more data/research?

> > > The blockchain sync failed for a 4th time. This is after removing the partition and downloading a new version of the blockchain on my main PC. I'm about ready to give up and go back to running the Stadicus version, so frustrating...
> > > However, on a brighter note, I was able to successfully unmount and check the HDD. I got the following output
> > 
> > 
> > Sorry to read that. Your fsck result looks OK.
> > What errors the blockchain sync failed with?
> > It can be also a network problem with the connection going down while transferring.
> > The Stadicus guide (on which the RaspiBlitz is based) has the exact same method for transferring the blockchain, just more manual.
> > If there is a problem with the scp transfer failing -because of an unstable network connection- you could still try downloading by the torrent method. Why not to give it a try?
> 
> I don't remember what the error was for the blockchain sync, but I went ahead and downloaded the torrent which worked without any issues. I'm still puzzled by why the "copy" with WinSCP didn't work even after reformatting the HD and downloading a new copy of the blockchain. Oh well, at least I'm up and running now. Appreciate the help.

SCP scp does not guarantee file integrity since it does not use file checksums like `rsync` does. So though unlikely it's possible you had some files corrupted when copying.

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issues/479#issuecomment-481335619_

rootzoll, 2019-04-09:
Since v1.1 RaspiBlitz already uses rsync instead of SCP for the Blockchain transfer process. But thats just if the user tells RaspiBlitz that the host compuer is OSX/Mac or Linux. Windows does not have rsync installed - so for Windows computers it defaults to SCP.

@fluidvoice Can I consider this issue closed, or is there anything to improve still?

fluidvoice, 2019-04-09:
Are there no client apps for Windows that talk rsync protocol?  There is cygwin, but that's prob a bit much to ask users to install?

fluidvoice, 2019-04-09:
[image]

rootzoll, 2019-04-09:
1st of all: I dont have a Windows machine to test or dev this - so this would need to be a PR. I add a "Help wanted" lable.

I am open to let the user install some additional software on windows as long as that one is free. Bit this mean for windows there is a whole other process to follow then for Mac/Linux. The script of 50copyHDD.sh has already a selection if user is running Mac/Linux or Windows. So that can be done. But having a GUI tool on the windows side would also mean there is not "one line of copy" to instruct the user just to copy and run. So there needs to be some kind of detailed documentation on how to install the tool and all the single clicks and selections to make. That is maybe a bit much to have all in the shell text window. So maybe a link to a youtube video or a FAQ text with pictures to follow?

fluidvoice, 2019-04-09:
> 1st of all: I dont have a Windows machine to test or dev this - so this would need to be a PR. I add a "Help wanted" lable.
> 
> I am open to let the user install some additional software on windows as long as that one is free. But this mean for windows there is a whole other process to follow then for Mac/Linux. The script of 50copyHDD.sh has already a selection if user is running Mac/Linux or Windows. So that can be done. But having a GUI tool on the windows side would also mean there is not "just one line of code" to instruct the user just to copy and run. So there needs to be some kind of detailed documentation on how to install the tool and all the single clicks and selections to make. That is maybe a bit much to have all in the shell text window. So maybe a link to a youtube video or a FAQ text with pictures to follow?

Makes sense.  Command line even on Windows is better.  Perhaps `RClone` is the tool to use. It would be nice if it works since we could use for backing up to cloud providers too. But someone needs to test it.
Note on Rclone...

> There are 3 main syncing methods
> 
> no flags - (size, modtime)
> --size-only (size)
> --checksum (size, checksum)
> Then there are the modifiers
> 
> --ignore-size makes all of the above skip the size check
> --ignore-times - uploads unconditionally (no checks)

Z-and, 2019-04-09:
I finally have my RaspiBlitz up and running with the torrent copy of the blockchain, so I don't want to start meddling with things again. However, I do have an extra Pi setup (Pi, microSD, HDD) that I could possibly test this out with. It would probably have to be later in the week though.

bitsam, 2019-04-24:
Is it documented anywhere to move the torrent files from one computer to a Raspiblitz node? I have the torrents seeding on a Windows computer and want to copy it locally to the Raspiblitz node using Rclone.

fluidvoice, 2019-04-24:
> Is it documented anywhere to move the torrent files from one computer to a Raspiblitz node? I have the torrents seeding on a Windows computer and want to copy it locally to the Raspiblitz node using Rclone.

Did you search the main Raspiblitz page for words like "blockchain" and "copy"? your answer is there.

bitsam, 2019-04-25:
All the instructions I can see refer to copying over a fully sync'd
blockchain, not just the torrent files. An explanation on where the torrent
files should be copied so the node thinks it has performed the torrent.

rootzoll, 2019-04-28:
@bitsam I added your question to the FAQs of v1.2 - see:
I have downloaded the torrents on another computer. How do I copy them to the RaspiBlitz?
https://github.com/rootzoll/raspiblitz/blob/v1.2/FAQ.md#i-have-downloaded-the-torrents-on-another-computer-how-do-i-copy-them-to-the-raspiblitz

_You need both torrents to be downloaded on your laptop for that. You need merge both torrents files now to a full blockchain copy. Stop torrent app. Copy all downloaded data from update torrent into the directory of the base torrent. Not you have a copy of the full blockchain in one directory and you can continue like in the FAQ case above "I have the full blockchain on another computer. How do I copy it to the RaspiBlitz?"_

nyxnor, 2021-05-17:
Update:
Blockchain is using rsync
Lnd backup is using scp

Can this be close or something else to add to this discussion?

rootzoll, 2021-05-18:
Closing. If still an issue or need for discussion - please reopen or comment.

closed 2021-05-18 by rootzoll