raspiblitz/raspiblitz #503: Update README and FAQ

rootzoll, 2019-04-09:
- update main image to raspiblitz with new heatsink-case and v1.2 on display
- add assemble info for new case (with images) to intructions
- add changes from v1.1 to v1.2 to FAQ
- add new download links for sd card image and torrent

holuan, 2019-04-10:
Also recovery part of FAQ needs some updating. I just had an sd card failure today and my raspiblitz was not accessible via SSH. FAQ suggests that my only option was restore from seed. However, lnd stores information on ssd so I was able to reflash the sd card and restore node fully.

So this part is technically misleading:

> To recover all your LND data you must still be able to SSH into the RaspiBlitz (minimum v1.1) and the HDD should be still useable/reachable (mounted) - even it shows some errors. If this is not possible anymore you should skip to the second option "Recover from Wallet Seed".

If you still can SSH in and HDD is readable, we can try to rescue/export your LND data (funds and channels) from a RaspiBlitz to then be able to restore it back to a fresh one. For this you can use the following procedure ...

>

fluidvoice, 2019-04-27:
https://github.com/rootzoll/raspiblitz/issues/550#issue-437978296

rootzoll, 2019-04-28:
TODO: Update Download Links v1.2 with SHA-SUM and torrent.

rootzoll, 2019-04-28:
@holuan thanks - I added your input to the v1.2 FAQ

closed 2019-06-23 by rootzoll