raspiblitz/raspiblitz #504: Add new Torrent Update

rootzoll, 2019-04-09:
The last torrent blockchain update file is more then 3 month old - needs update.

rootzoll, 2019-04-09:
Added a new update torrent - needs testing

rootzoll, 2019-04-09:
TODO: When local testing with torrent was positive - ask community over twitter to help seeding the torrent.

kilrau, 2019-04-11:
My raspiblitz is sitting on a 1Gbit upload connection here and I would be happy to help with seeding. Can I use the raspi-internal torrent client for this? If so, please point me to some instructions.

rootzoll, 2019-04-11:
@kilrau moved the torrent seeding feature to the v1.2 release .. you can subscribe to this issue, to get an info on updates: https://github.com/rootzoll/raspiblitz/issues/24

kilrau, 2019-04-11:
Done, thx!

rootzoll, 2019-04-11:
OK - new Torrent Update File is tested and works. Opening a new issue to ask people seeding. Closing this issue.

closed 2019-04-11 by rootzoll