raspiblitz/raspiblitz #506: Unable to resolve host after update to 1.1 from 1.0

lucbran, 2019-04-10:
Hello
I am updating version but system locked as unable to resolve host..
Strangely config file shows version 0.98 while I have followed the process in updating to 1.1:
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='0.98'
hostname=blitz
network=bitcoin
chain=main
autoPilot=on
autoNatDiscovery=off
runBehindTor=off
rtlWebinterface=on
publicIP=82.102.21.52
autoUnlock=off

can anyone help? thanks

bhavman2, 2019-04-11:
I had a similar issue [#462](https://github.com/rootzoll/raspiblitz/issues/462)

Ended up solving the problem by correcting the hostname in the following locations:

sudo nano /etc/hostname
[image]

sudo nano /etc/hosts
[image]

Not sure about the config file showing the wrong version.

lucbran, 2019-04-12:
Thanks I also solved issue: changing  hostname was not enough so I changed it, burnt SD card and reinstalled.. The updated hostname now shows well and working..

closed 2019-04-12 by lucbran

fluidvoice, 2019-04-12:
> so I changed it, burnt SD card and reinstalled.

"changed it"... you changed what?

lucbran, 2019-04-12:
Ok I understand the problem was due to the fact I changed alias name (BRNode) using asterisks (**BRNode**) Same was probably not accepted and generating error , so I changed alias name into BRNode (no asterisks)  from sudo nano /mnt/hdd/raspiblitz.conf but when rebooting same not working..
So I tried after reinstate original name BRNode from sudo nano /mnt/hdd/raspiblitz.conf and reburnt the SD card.
After that sys recognised the original Alias name and the whole process of updating to 1.1 version was finalized..
Hope clear..