raspiblitz/raspiblitz #508: Power Cord from Amazon on USA list "Currenly out of Stock"

ryanortega, 2019-04-10:
Currently out out of stock:  Power: http://a.co/d/fVVOYZZ

Should it be replaced with another one for now?  Possibly: https://www.amazon.com/dp/B01N336XEU/ref=dp_cerb_3

fluidvoice, 2019-04-10:
Yeah it looks pretty good. From the Amazon reviews...

> Take note that I'm talking specifically about the >>> Pi 3 B+ <<<, not the Pi 3 or even the Pi 3B. I'm also talking about powering both the touchscreen and the Pi 3 B+ from a single wart. With one wart powering both the measured voltage dropped down to 5.1 volts and I had the low voltage warning on the display. I tested the Pi 3 B+ without the touchscreen connected and there was no warning. The voltage measured 5.22 volts without the screen. 

[image]

rootzoll, 2019-04-11:
@ryanortega I would highly recommend this one: https://www.amazon.com/dp/B07MC7B9X3

rootzoll, 2019-04-11:
USA Shopping List Updated. Closing the issue.

closed 2019-04-11 by rootzoll