raspiblitz/raspiblitz #510: Flashed SD Card, Expected Previous State But Now Asking To Start From The Beginnning

funk256, 2019-04-10:
I had problems previously and flashed the SD card again.  That did the trick to fix it at that time.  Of course it prompts you for the standard raspiblitz password at first then you change it to whatever you want, go through to the next screen, and it will sync back up and all of that.

I had problems again with the device booting and being visible on the network.  So I decided to flash the card again and this time instead of giving me any indication of prior data or anything saved on the HDD it's asking me to start from scratch.  Typing in raspiblitz as the password takes you to the first screen where you choose between being a Bitcoin or Litecoin node.  Then it prompts you to save your four passwords again but I just decided to exit once it got to that screen.  Oh, it's also giving me a new IP address if that helps one way or the other.  Used to end in 111 but now it's 122.

Any idea why that's happening this time around?  As best I can tell I didn't change anything on my computer, with the HDD, or the SD card itself outside of flashing it again to see if I could get it responsive.  Which is certainly is now responsive but, of course, if I have to start from the beginning that isn't much help and I honestly think I'd probably just tap out and not bother with it again.

Thanks to anyone that can give some input.

fluidvoice, 2019-04-11:
> I had problems again with the device booting

Did you ever upgrade to a better SD card as I recommended in your previous issue?
You may now have also had data corruption on your HDD where your blockchain and LND data is stored.
If so, it could be a result of an inadequate power suppl - which is known that it *can* cause file corruption if infrequently.  Run this power-test script from SSH command line on your node (all one line):

`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/4abbdd18aa0ec44553d1fd875f2c5a6b98d9721c/home.admin/config.scripts/blitz.powertest.sh && bash ./blitz.powertest.sh`

funk256, 2019-04-11:
Yeah, I was just leaving it up and checking it every so often to see how it was doing.  Hard to recall now but it must have been a week or close to two with no hiccups.  Then I came home a few days ago and it was off.  I tried to get it to come on and I never could discover it with Advanced IP Scanner (never used the screen, I just get the IP from there and SSH in).  I have another SD card I can, and probably will at this rate, try this weekend.  It's newer but maybe cheaper at the same time so can't say I'm too well versed on the quality of one versus the other but certainly can't hurt just to try another.  We had a storm while I was away at work so I was wondering if that kicked it off or maybe caused some sort of corruption but I'm just speculating now.

I am using this power supply:  https://www.amazon.com/gp/product/B07HSPT6MG/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1  I also grabbed a different case.  Otherwise I should have everything exactly as described in the guide.  I grabbed those to get two day shipping as the others would be delayed but perhaps that was a mistake.

In any case, the message below is what I get from that command.  The last line kept looping and looping for probably a total of 20 times before I exited.  Thanks again for all of your help.

admin@RaspiBlitz:~ $ wget https://raw.githubusercontent.com/rootzoll/raspiblitz/                                                                                                                                                             4abbdd18aa0ec44553d1fd875f2c5a6b98d9721c/home.admin/config.scripts/blitz.powerte                                                                                                                                                             st.sh && bash ./blitz.powertest.sh
--2019-04-11 01:30:58--  https://raw.githubusercontent.com/rootzoll/raspiblitz/4                                                                                                                                                             abbdd18aa0ec44553d1fd875f2c5a6b98d9721c/home.admin/config.scripts/blitz.powertes                                                                                                                                                             t.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.56.13                                                                                                                                                             3
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.56.1                                                                                                                                                             33|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 821 [text/plain]
Saving to: ‘blitz.powertest.sh’

blitz.powertest.sh  100%[===================>]     821  --.-KB/s    in 0s

2019-04-11 01:30:59 (7.18 MB/s) - ‘blitz.powertest.sh’ saved [821/821]

VCHI initialization failed 1400/   0 MHz VCHI initialization failed
VCHI initialization failed  600/   0 MHz VCHI initialization failed
VCHI initialization failed  600/   0 MHz VCHI initialization failed

fluidvoice, 2019-04-11:
SD cards are listed in the shopping lists as are power supply recommendations. That's a good place to start.  Looks like perhaps that power test did not run correctly.

fluidvoice, 2019-04-11:
The "official" Pi PS is this one: https://www.newark.com/raspberry-pi/t5875dv/psu-raspberry-pi-5v-2-5a-multi/dp/77Y6535

fluidvoice, 2019-04-11:
also you might get that other test to run with this:
`chmod +x blitz.powertest.sh && sudo ./blitz.powertest.sh`

funk256, 2019-04-11:
That's fair.  I may go that route.  Will have to try some other stuff and think about it a bit.  Not sure if something got cut off there but results of the first line you suggested to try:

bmariadbclient18 armhf 10.1.37-0+deb9u1 [714 kB]
Get:3 http://raspbian-us.ngc292.space/raspbian stretch/main armhf sysbench armhf                                                                                                                                                              0.4.12-1.2 [55.8 kB]
Fetched 776 kB in 2s (284 kB/s)
Selecting previously unselected package mysql-common.
(Reading database ... 81392 files and directories currently installed.)
Preparing to unpack .../mysql-common_5.8+1.0.2_all.deb ...
Unpacking mysql-common (5.8+1.0.2) ...
Selecting previously unselected package libmariadbclient18:armhf.
Preparing to unpack .../libmariadbclient18_10.1.37-0+deb9u1_armhf.deb ...
Unpacking libmariadbclient18:armhf (10.1.37-0+deb9u1) ...
Selecting previously unselected package sysbench.
Preparing to unpack .../sysbench_0.4.12-1.2_armhf.deb ...
Unpacking sysbench (0.4.12-1.2) ...
Setting up mysql-common (5.8+1.0.2) ...
update-alternatives: using /etc/mysql/my.cnf.fallback to provide /etc/mysql/my.c                                                                                                                                                             nf (my.cnf) in auto mode
Setting up libmariadbclient18:armhf (10.1.37-0+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up sysbench (0.4.12-1.2) ...

***
RaspiBlitz powertest v0.1
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
***

sysbench 0.4.12:  multi-threaded system evaluation benchmark

Running the test with following options:
Number of threads: 4

Doing CPU performance benchmark

Threads started!
59.1'C 1400 MHz  1.2750V
63.4'C 1200/1400 MHz 1.2750V
64.5'C 1200/1400 MHz 1.2750V
65.5'C 1200/1400 MHz 1.2750V
66.1'C 1200/1400 MHz 1.2750V
66.6'C 1200/1400 MHz 1.2750V
67.7'C 1200/1400 MHz 1.2750V
Done.

Maximum prime number checked in CPU test: 10000

Test execution summary:
    total time:                          35.5980s
    total number of events:              10000
    total time taken by event execution: 142.3535
    per-request statistics:
         min:                                 11.94ms
         avg:                                 14.24ms
         max:                                 36.89ms
         approx.  95 percentile:              15.70ms

Threads fairness:
    events (avg/stddev):           2500.0000/6.36
    execution time (avg/stddev):   35.5884/0.00

63.4'C 1200/1400 MHz 1.2750V
62.3'C  600/1400 MHz 1.2750V
61.8'C  600/1400 MHz 1.2750V
60.7'C  600/1400 MHz 1.2750V
60.7'C  600/1400 MHz 1.2750V
60.1'C  600/ 600 MHz 1.2750V
59.1'C  600/1400 MHz 1.2750V
59.1'C  600/1400 MHz 1.3750V

Results from the second command (guessing that keeps running so just copied the top five lines):

admin@RaspiBlitz:~ $ chmod +x blitz.powertest.sh && sudo ./blitz.powertest.sh
56.4'C  600/ 600 MHz 1.2V
56.4'C  600/ 600 MHz 1.2V
56.4'C  600/ 600 MHz 1.2V
56.9'C  600/ 600 MHz 1.2V
56.4'C  600/ 600 MHz 1.2V

The HDD isn't flashing.  I just have that connected and a monitor which is currently powered down.  Are you thinking my HDD might be corrupted?  I would assume there is no way back from that.  Certainly not the end of the world though a little disappointing if so.  It is what it is, no worries.  Thanks again.

fluidvoice, 2019-04-11:
> admin@RaspiBlitz:~ $ chmod +x blitz.powertest.sh && sudo ./blitz.powertest.sh
> 56.4'C 600/ 600 MHz 1.2V
> 56.4'C 600/ 600 MHz 1.2V
> 56.4'C 600/ 600 MHz 1.2V
> 56.9'C 600/ 600 MHz 1.2V
> 56.4'C 600/ 600 MHz 1.2V

Yeah, the power supply looks weak. This is a lot of throttling (to 600MHz) due to only 1.2 volts when the CPU should be running at 1400Mhz.

fluidvoice, 2019-04-11:
> That's fair. I may go that route. Will have to try some other stuff and think about it a bit. Not sure if something got cut off there but results of the first line you suggested to try: [...]

@openoms what's the difference between your test and the one @rootzoll made?  It seems yours is showing a lower value of 1.2V.  Might have relevance since I think @rootzoll set his scripts to warn at 1.25V or below.

funk256, 2019-04-11:
Okay, no problem. May end up getting a new one. Thanks again.

rootzoll, 2019-04-11:
@fluidvoice thanks for helping out. I hope to get v1.2 out of the door next week. It will have the powertest at the beginning of setup and lots more to prevent those data corruption cases.

I will close this issue for now.

closed 2019-04-11 by rootzoll