raspiblitz/raspiblitz #511: Keep getting "[lncli] rpc error: code = Unknown desc = peer disconnected" when attempting to open channels with certain nodes

Z-and, 2019-04-11:
I've encountered this issue with about half the nodes I tried to open channels with. I can connect as peer without an issue but whenever I open a channel it stalls then eventually gives me the following output.

`[lncli] rpc error: code = Unknown desc = peer disconnected`

Below is my node status, everything looks to be okay. Not sure why this keeps happening.

```
    "identity_pubkey": "032f0d2d9f9b14b3a40488a1f2f255ecfd73b2272ae6d8399cbf067b0b55913281",
    "alias": "Zand",
    "num_pending_channels": 0,
    "num_active_channels": 5,
    "num_peers": 8,
    "block_height": 571088,
    "block_hash": "0000000000000000001d96002b130121b32b5b1d91bedbd637ae9900f1d9ce76",
    "synced_to_chain": true,
    "testnet": false,
    "chains": [
        "bitcoin"
    ],
    "uris": [
        "032f0d2d9f9b14b3a40488a1f2f255ecfd73b2272ae6d8399cbf067b0b55913281@173.52.56.208:9735"
    ],
    "best_header_timestamp": "1554931014",
    "version": "0.5.2-beta commit=v0.5.2-beta",
    "num_inactive_channels": 1
```

CommanderPoe, 2019-04-11:
Same happened yesterday to me with a friend, then i went and tried with few other nodes and worked perfectly... I think it has to do something with the ports of the other node, not yours.

Z-and, 2019-04-11:
> Same happened yesterday to me with a friend, then i went and tried with few other nodes and worked perfectly... I think it has to do something with the ports of the other node, not yours.

I don't think that's the case here, the nodes I'm having issues with are very well connected/funded nodes (LNBIG, rompert, tippin.me etc..)

Z-and, 2019-04-12:
So I checked some logs and found the following errors, not sure if this could relate to the issue but figured I'd share (the same error pops up numerous times with different IP addresses)

`2019-04-12 21:53:45.118 [INF] RPCS: [listchannels] fetched 9 channels from DB
2019-04-12 21:53:49.194 [ERR] CMGR: Can't accept connection: unable to accept co                                                                                                                                                             nnection from 195.201.208.22:36722: chacha20poly1305: message authentication fai                                                                                                                                                             led
2019-04-12 21:53:56.958 [INF] DISC: Broadcasting batch of 109 new announcements
2019-04-12 21:53:57.010 [ERR] CMGR: Can't accept connection: unable to accept co                                                                                                                                                             nnection from 173.52.56.208:1024: EOF`

fluidvoice, 2019-04-15:
> I don't think that's the case here, the nodes I'm having issues with are very well connected/funded nodes (LNBIG, rompert, tippin.me etc..)

well connected/funded doesn't mean they are running the latest code. If your problem is specific to only some nodes and doesn't happen on all of them then it could be they are running old code?

Z-and, 2019-04-16:
> well connected/funded doesn't mean they are running the latest code. If your problem is specific to only some nodes and doesn't happen on all of them then it could be they are running old code?

I believe I found the source of the problem. I noticed opening port 22 on the Pi can really affect performance and connectivity (not to mention security). I had it open so I could ssh remotely from a computer outside my local network. As soon as I closed port 22 on my firewall, I was able to connect to the nodes I previously had issues with.

closed 2019-04-16 by Z-and