raspiblitz/raspiblitz #513: Help Torrent Seeding for RaspiBlitz v1.2 Release

rootzoll, 2019-04-11:
Hi there. 

Much thanks to everybody helped on seeding the RaspiBlitz blockchain so far :)
Welcome if you are new volunteer 👍 

In the preparation of the upcomming RaspiBlitz v1.2 release there will be a torrent file for the blockchain update that needs good seeding ...

**Most important is to seed the new UPDATE torrent:**
https://github.com/rootzoll/raspiblitz/raw/v1.2/home.admin/assets/raspiblitz-bitcoin1-2019-04-09-update.torrent

**But also people see some slow downloads of the BASE torrent - so please add to seeding also if you are not seeding already:**
https://github.com/rootzoll/raspiblitz/raw/v1.2/home.admin/assets/raspiblitz-bitcoin1-2018-10-13-base.torrent

vamp111, 2019-04-11:
I'm on it!

bhavman2, 2019-04-11:
Trying to DL now, will seed after!

kilrau, 2019-04-11:
[image]

Will let it seed on this machine until https://github.com/rootzoll/raspiblitz/issues/24 is working.

pxsocs, 2019-04-12:
+1. Seeding.

bitsam, 2019-04-12:
what torrent client do you recommend for Windows ? I noticed a lot of Libtorrent clients when downloading the blockchain.

thelwyn, 2019-04-12:
On Windows, I use qBittorrent , works fine.

mamar59, 2019-04-12:
Hi rootzoll I share the torrents to participate in the deploment raspiblitz, I have a good upload connection in France. do you think I need to share the file too https://github.com/rootzoll/raspiblitz/raw/v1.2/home.admin/assets/raspiblitz-bitcoin1-2018-10-13-base.torrent I have the same date on another file

[image]

rootzoll, 2019-04-12:
@mamar59 thats the same torrent file (just in another branch of the GitHub) - you are already seeding all of the important files - much thanks :D

philbertw4, 2019-04-16:
I'm not sure if what I'm about the propose is a good idea or not (or if it has been proposed earlier), but:
* since raspiblitz has an option for downloading the blockchain via torrent as part of its setup process
* perhaps we consider adding "seed the raspiblitz blockchain torrent" as a service on the raspiblitz itself?
* perhaps by default it should be turned on but, as a security precaution, users must confirm this setting prior to completing setup?

Then again I'm not an expert in torrenting or raspberry-pi-ing, so maybe this is too much overhead for the small device and not very helpful for the purpose of seeding the torrent?

rootzoll, 2019-04-17:
OK I think we have a good basic seed for the upcomming v1.2 release :)
Stay tuned. Closing Issue.

closed 2019-04-17 by rootzoll

kilrau, 2019-04-17:
I think you should also ask to seed the 1.2 image itself once it's out.