raspiblitz/raspiblitz #514: When using "sudo journalctl -f -u lnd" to check logs nothing appears

Z-and, 2019-04-11:
When I check my logs with "sudo journalctl -f -u lnd" command, nothing seems to show up (even if I open a separate SSH window and connect to a peer or open a channel). I just see the following

```
admin@Zand:~ $ sudo journalctl -f -u lnd
-- Logs begin at Thu 2019-04-11 16:51:42 BST. --

```

rootzoll, 2019-04-12:
The journal logs were reduced basically to ERROR level - but normally you should see at least something like:
```
Stopped LND Lightning Daemon.
Started LND Lightning Daemon.
```
For more details on logs, run the script: /home/admin/XXdebugLogs.sh

closed 2019-04-12 by rootzoll

Z-and, 2019-04-12:
> /home/admin/XXdebugLogs.sh

Great thank you!

Z-and, 2019-04-12:
> The journal logs were reduced basically to ERROR level, because journal logs get written to the /var/logs on the sd card and we wanted to reduce the writing load on the sd card.
> 
> Bbut normally you should see at least something like:
> 
> ```
> Stopped LND Lightning Daemon.
> Started LND Lightning Daemon.
> ```
> But there is still more detailed LND logging ... For more details on logs, run the script: `/home/admin/XXdebugLogs.sh`

Great thank you!