raspiblitz/raspiblitz #516: No HDD found when doing upgrade to new version

joelstahre, 2019-04-12:
I flashed the new version onto the SD card.

When i put the SD card into the PI and i ssh into the machine with default username and password it says that no partition exists on the HDD, and im asked if i want to create one.

I suspect this is not normal.

openoms, 2019-04-12:
some more info would help. Did you use the HDD before? Is there anything on it?  Did you run a previous version of the RaspiBlitz?

joelstahre, 2019-04-12:
Sorry, yes i did run the previous version, so the HDD has been working fine with the old version.

fluidvoice, 2019-04-12:
ssh in and run `lsblk` on the command line and post the output here.

joelstahre, 2019-04-12:
> what was the message you saw, was it?...
> "WARNING: NO HDD FOUND -> Shutdown, connect HDD and restart."
> 
> SSH in and run this on the command line and post the output here:
> `sudo cat /etc/fstab && lsblk && blkid |grep /dev/sd && df -h`

`admin@RaspiBlitz:~ $ sudo cat /etc/fstab && lsblk && blkid |grep /dev/sd && df -h
proc            /proc           proc    defaults          0       0
PARTUUID=03d9b017-01  /boot           vfat    defaults          0       2
PARTUUID=03d9b017-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk
mmcblk0     179:0    0  14.9G  0 disk
|-mmcblk0p1 179:1    0  43.9M  0 part /boot
`-mmcblk0p2 179:2    0   7.3G  0 part /`

fluidvoice, 2019-04-12:
Your HDD is missing the partition sda1.  What does your raspiblitz.log have in it?...
`cat /home/admin/raspiblitz.log`

fluidvoice, 2019-04-12:
> Your HDD is missing the partition sda1. What does your raspiblitz.log have in it?...
> `cat /home/admin/raspiblitz.log`

@rootzoll For debugging, if (**IF**) the scripts delete any partitions it should should be echoed into a log file that survives between SD card rewriting.  So eg., if this is logged to /home/admin/raspiblitz.log then perhaps that log file should be moved to /mnt/hdd

joelstahre, 2019-04-12:
```admin@RaspiBlitz:~ $ cat /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.1
Thu 11 Apr 20:12:49 BST 2019
***********************************************```

fluidvoice, 2019-04-12:
wow, that doesn't show anything. I was hoping it would at least show whatever might have been done to the HDD, if anything.  I'm not sure how/why your HDD lost that partition.

joelstahre, 2019-04-12:
hehe no :/ i guess i just have to redo the whole thing again.

CommanderPoe, 2019-04-15:
> wow, that doesn't show anything. I was hoping it would at least show whatever might have been done to the HDD, if anything. I'm not sure how/why your HDD lost that partition.

Werent all logs removed after version 1.1 due to the amount space they were using?

fluidvoice, 2019-04-15:
> > wow, that doesn't show anything. I was hoping it would at least show whatever might have been done to the HDD, if anything. I'm not sure how/why your HDD lost that partition.
> 
> Werent all logs removed after version 1.1 due to the amount space they were using?

system logs yes. Raspiblitz, Bitcoin, and LND logs are not removed.

rootzoll, 2019-04-16:
For v1.2 I added a confirmation dialog before HDD gets formatted. So just in case something is not detected correct on upgrade and it switches to fresh setup - the user has to confirm to format HDD.

TODO: final testing

rootzoll, 2019-04-27:
Worked so far for v1.2 - closing issue.

_If there is still some "HDD not found while setup but there is one connected" - please open in new issue. Thx._

closed 2019-04-27 by rootzoll