raspiblitz/raspiblitz #518: Unable to ssh while final blockchain sync

bitsam, 2019-04-12:
I have a new setup and up to the final blockchain synchronisation. at around 95.69% . I leave an SSH window open on a NAS server and overnight the connection drops and then I am unable to SSH back onto the raspiblitz node. This happened yesterday and I restarted the node and it went back to 95.56%. 
Should I just leave it running, the node itself looks normal with LAN led's blinking, and the ocasional raspi LED flashing.

bitsam, 2019-04-13:
Can we exit out of the blue screen and start the Blockchain sync from command line? It would be nice to see what is happening, what speed is it downloading etc... This is after the torrent and performing the final sync. Also can someone tell me how to start up the Wizard / Blue window after you have exited to the command prompt ?

fluidvoice, 2019-04-14:
> Can we exit out of the blue screen and start the Blockchain sync from command line?

Yes.  In SSH hit Ctrl-c to escape to the command line. Then run:
`sudo systemctl stop lnd && sudo systemctl stop bitcoind`
`sudo /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf`
`sudo tail -f /mnt/hdd/bitcoin/debug.log`

fluidvoice, 2019-04-14:
> then it seems the node hangs.

what does this mean? what do you see?  Screen pics would help.

bitsam, 2019-04-15:
Thanks @fluidvoice  running the sync in cmd is much more stable. The hang is hard to troubleshoot. I did get the occasional under voltage error, and then occasionally I had another error pop up on the LCD, but it was too quick for me to snap a photo of. Apart from that the node looked fine from the LCD. I did notice that the HDD activity seemed to stop after about an hour. So I would re-boot, and continue with the wizard, adn then about 1 hr later I could no longer SSH back onto the node, HDD activity would stop, but the LCD remained on the "Login to your RaspiBlitz with Password A on x.x.x.x" . I also tried disconnecting the LAN cable and the LCD would change, detecting the loss of network, and then detect when i connected the network again. However I still could not SSH.

However after running the commands below, I have been able to continue syncing the blockchain at around 1% a day. (97% @ present)

> sudo systemctl stop lnd && sudo systemctl stop bitcoind
sudo /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf
sudo tail -f /mnt/hdd/bitcoin/debug.log
> 

Here is the export of  ./XXdebugLogs.sh

*** RASPIBLITZ LOGS ***
blitzversion: 1.1
chainnetwork: bitcoin / main
 21:26:44 up 22:54,  2 users,  load average: 2.68, 3.09, 3.12

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: failed (Result: timeout) since Sun 2019-04-14 22:42:32 BST; 22h ago
  Process: 740 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 759 (code=killed, signal=KILL)

Apr 14 22:42:32 RaspiBlitz systemd[1]: bitcoind.service: Unit entered failed state.
Apr 14 22:42:32 RaspiBlitz systemd[1]: bitcoind.service: Failed with result 'timeout'.

*** LAST 5 ERROR LOGS ***
-- Logs begin at Sun 2019-04-14 22:17:01 BST, end at Mon 2019-04-15 21:26:44 BST. --
Apr 14 22:42:31 RaspiBlitz systemd[1]: bitcoind.service: Killing process 759 (bitcoind) with signal SIGKILL.
Apr 14 22:42:32 RaspiBlitz systemd[1]: bitcoind.service: Main process exited, code=killed, status=9/KILL
Apr 14 22:42:32 RaspiBlitz systemd[1]: Stopped Bitcoin daemon.
Apr 14 22:42:32 RaspiBlitz systemd[1]: bitcoind.service: Unit entered failed state.
Apr 14 22:42:32 RaspiBlitz systemd[1]: bitcoind.service: Failed with result 'timeout'.

*** LAST 20 INFO LOGS ***
2019-04-15T20:22:46Z UpdateTip: new best=000000000000000000190c386d90068d0258ed88edaa85246a0c29e345bb8b49 height=563927 version=0x20000000 log2_work=90.368077 tx=384954612 date='2019-02-20T18:47:28Z' progress=0.971701 cache=65.0MiB(541860txo)
2019-04-15T20:22:46Z UpdateTip: new best=0000000000000000001bd780ad63d2969eb13d6eac8bb76a8afe1d384b743742 height=563928 version=0x20c00000 log2_work=90.368101 tx=384954613 date='2019-02-20T18:56:17Z' progress=0.971704 cache=65.0MiB(541861txo)
2019-04-15T20:23:02Z UpdateTip: new best=0000000000000000000f6d10463a35c0138933f16fc610c66c1551276ac843c7 height=563929 version=0x20c00000 log2_work=90.368124 tx=384957726 date='2019-02-20T19:31:53Z' progress=0.971717 cache=65.6MiB(547351txo)
2019-04-15T20:23:02Z Syncing txindex with block chain from height 208040
2019-04-15T20:23:16Z Pre-allocating up to position 0x600000 in rev01535.dat
2019-04-15T20:23:16Z UpdateTip: new best=000000000000000000235fa5e6ce63c50db42f6978a6576ba889e85b8736838b height=563930 version=0x3fff0000 log2_work=90.368148 tx=384960878 date='2019-02-20T19:42:42Z' progress=0.971721 cache=65.7MiB(548672txo)
2019-04-15T20:23:39Z UpdateTip: new best=00000000000000000014e2f61dbe8dee221e0df172990b62f2651ed4d7da0f17 height=563931 version=0x20000000 log2_work=90.368171 tx=384963200 date='2019-02-20T19:45:08Z' progress=0.971721 cache=66.1MiB(552485txo)
2019-04-15T20:23:51Z UpdateTip: new best=0000000000000000000b348ee3cfe241384a3b3923f9f6b3ea239ceb8c1f6619 height=563932 version=0x2000e000 log2_work=90.368195 tx=384966360 date='2019-02-20T20:02:43Z' progress=0.971728 cache=66.5MiB(556450txo)
2019-04-15T20:23:51Z Syncing txindex with block chain from height 208042
2019-04-15T20:24:11Z UpdateTip: new best=0000000000000000001e6cf2c6888c297d6ed612555cf936ea24b452d669f0b3 height=563933 version=0x20400000 log2_work=90.368218 tx=384969254 date='2019-02-20T20:10:15Z' progress=0.971731 cache=66.9MiB(560026txo)
2019-04-15T20:24:24Z UpdateTip: new best=000000000000000000124e4f94e0921f33335cdd217dfebaeb0a2ff126bb321c height=563934 version=0x20800000 log2_work=90.368242 tx=384972100 date='2019-02-20T20:20:23Z' progress=0.971734 cache=67.3MiB(563440txo)
2019-04-15T20:24:39Z UpdateTip: new best=00000000000000000024d762fb720529ec5efeea510011a82d53858f0fa4543d height=563935 version=0x3fffe000 log2_work=90.368265 tx=384975299 date='2019-02-20T20:39:13Z' progress=0.971741 cache=67.8MiB(568004txo)
2019-04-15T20:24:57Z Pre-allocating up to position 0x700000 in rev01535.dat
2019-04-15T20:24:57Z UpdateTip: new best=000000000000000000277f9eb5878056eb23badc33c43378de05af8b49078ef4 height=563936 version=0x20000000 log2_work=90.368289 tx=384978225 date='2019-02-20T20:42:58Z' progress=0.971743 cache=68.1MiB(571054txo)
2019-04-15T20:24:57Z Syncing txindex with block chain from height 208043
2019-04-15T20:25:25Z UpdateTip: new best=00000000000000000022f8601a6ab772f60bd26557cc74458044f98297ddf0c8 height=563937 version=0x20c00000 log2_work=90.368312 tx=384980893 date='2019-02-20T20:45:03Z' progress=0.97 [... cut ...]
2019-04-14 22:19:46.889 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-04-14 22:19:46.934 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-04-14 22:19:46.934 [INF] RPCS: password RPC server listening on [::]:10009
2019-04-14 22:19:46.935 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config

fluidvoice, 2019-04-15:
> I did get the occasional under voltage error, and then occasionally I had another error pop up on the LCD

Which power supply are you using?  I advise you stop what you're doing and go buy a new/better power supply. One of those listed in the shopping list guides.  Bad power could cause corruption of your HDD and blockchain and then you will have wasted a lot of time syncing the blockchain.  While you're at it, you should make sure the SD card you are using is also a good one (from the shopping lists).

bitsam, 2019-04-18:
@fluidvoice I am using a 3Amp almost identical to the one on the shopping list (as the model in the shopping list did not ship to my country). I was using this on a power board which also had a high amp multi USB charger. So I moved the power supply to a standalone wall outlet. I also just re-imaged the SD card which is a 16Gb Sandisk Ultra SDHC. I also ran a bad block scan and completely re-formatted it. I think I had an issue writing the initial image. The system is now much more stable, I don't seem to get the freezing, and have been able to ssh after many hours of uptime.

You can close the issue now. Thanks @fluidvoice for your help.

closed 2019-04-18 by bitsam