raspiblitz/raspiblitz #520: raspiblitz shows 0 balance sometimes but works normally other times

philbertw4, 2019-04-13:
After getting my raspiblitz up and running, everything appeared to be working normally so I proceeded to fund it the on-chain wallet. Even after the funding transaction had received over 10 confirmations it was not showing up in LND as as available on-chain balance, however it did show the transaction in the transaction history. 

This was strange behavior, so I then rebooted the whole thing. After reboot everything seemed to work normally. It showed the balance and even let me open a channel. I opened one manually (autopilot, while on, did not seem to be doing anything so I opened on on my own).

Then, while waiting for the channel funding transaction to confirm, I rebooted (I don't remember why -- thinking it was because I turned the RTL web interface on in the Services menu, or the auto-unlock feature). After reboot, everything loaded up and was operational again (seemingly) yet, strangely, my on-chain balance was zero again, nor did it show any information about the channel that I was opening. 

Rebooted again and now everything seems to be normal (eg it shows the remaining on-chain balance while I wait for the channel to open). 

I don't know how to begin diagnosing this issue, please let me know if there are specific commands that would be helpful for me to run.

fluidvoice, 2019-04-15:
Things change slowly sometimes due to network data transfers.  Rebooting will change how and in what order things happen.  If you don't still have a problem please close this issue.

philbertw4, 2019-04-16:
Thanks for your reply.

Yeah, the strange part (and why I opened the issue) was that it was giving this behavior even after the funding of the on-chain wallet (before even attempting to open lightning channels) had been confirmed by over 10 blocks. Regardless, as you suspected, the issue seems to have resolved now on its own. 

It's unfortunate because the user experience is certainly degraded and confusing when the node appears to behave unexpectedly. 

I intend to incubate some more raspiblitz nodes and will report back if the same thing happens again and it doesn't appear to be not due to network data transfers. Right now thought it appears that you're correct and that was the cause.

Closing this issue for now.

closed 2019-04-16 by philbertw4