raspiblitz/raspiblitz #525: Opening up port 22 on the Pi seems to cause connectivity/performance issues

Z-and, 2019-04-16:
> > > Same happened yesterday to me with a friend, then i went and tried with few other nodes and worked perfectly... I think it has to do something with the ports of the other node, not yours.
> > 
> > 
> > I don't think that's the case here, the nodes I'm having issues with are very well connected/funded nodes (LNBIG, rompert, tippin.me etc..)
> 
> well connected/funded doesn't mean they are running the latest code. If your problem is specific to only some nodes and doesn't happen on all of them then it could be they are running old code?

I believe I found the source of the problem. I noticed this weekend opening port 22 on the Pi can really affect performance and connectivity (not to mention security). I had it open so I could ssh remotely from a computer outside my local network. As soon as I closed port 22 on my firewall, I was able to connect to the nodes I previously had issues with. I'm wondering if opening port 3000 to use RTL remotely will cause similar issues, will have to test that out as well.

_Originally posted by @Z-and in https://github.com/rootzoll/raspiblitz/issues/511#issuecomment-483734699_

fluidvoice, 2019-04-16:
What exactly is your "firewall"?

Z-and, 2019-04-16:
> What exactly is your "firewall"?

What do you mean what exactly is it? I use Verizon as an ISP and their router has a firewall

fluidvoice, 2019-04-16:
> > What exactly is your "firewall"?
> 
> What do you mean what exactly is it? I use Verizon as an ISP and their router has a firewall

A firewall can be software running on your PC (eg. Windows has a firewall) or it can be in a WiFi or other router.  What is the model of the Verizon router?  What happens if you open port 22 again, does the problem come back?

Z-and, 2019-04-16:
The router model is a Actiontec MI424WR. It’s not an issue for me to keep port 22 closed, I can still use my node while at home or remotely through Zap wallet. I’m simply pointing out that keeping port 22 open seems to cause issues.

fluidvoice, 2019-04-16:
> The router model is a Actiontec MI424WR. It’s not an issue for me to keep port 22 closed, I can still use my node while at home or remotely through Zap wallet. I’m simply pointing out that keeping port 22 open seems to cause issues.

perhaps yes - with **your** router and/or ISP.  I don't think this is a problem for everyone else.
But it's good to report whatever problems you have, just en case others hit the same problem.
And most probably should not allow SSH from the internet as it's a potential security hole.

nyxnor, 2021-05-17:
Please close this.

closed 2021-05-18 by rootzoll